arkypita/LaserGRBL #86: GRBL comments on $$ settings

jorgemarmo, 2017-08-30:
this is more of a clarifying question, how do LaserGRBL handles the comments in the $$ settings? because I just read this:
----------------------------------------------------
_Grbl Settings

$$ - View Grbl settings

To view the settings, type $$ and press enter after connecting to Grbl. Grbl should respond with a list of the current system settings, as shown in the example below. All of these settings are persistent and kept in EEPROM, so if you power down, these will be loaded back up the next time you power up your Arduino.

The x of $x=val indicates a particular setting, while val is the setting value. In prior versions of Grbl, each setting had a description next to it in () parentheses, but **Grbl v1.1+ no longer includes them unfortunately. This was done to free up precious flash memory** to add the new features available in v1.1. However, most good GUIs will help out by attaching descriptions for you, so you know what you are looking at._
from [https://github.com/gnea/grbl/wiki/Grbl-v1.1-Configuration](url)
----------------------------------------------------
but when I save / load a configuration file generated by LaserGRBL (*.nc), the comments are still there, so are comments added to the config file and omitted when sending them to the arduino?
OR
most likely this comments never go to EEPROM even if you send them cuz () are omitted.
This whole thing come after I got errors and the final 4~5 settings where not updated when importing a previously generated configuration. after trying several times (about 4) it all went well.
probably a longer delay between each line could help? ....

arkypita, 2017-08-30:
Dear @jorgemarmo 
LaserGRBL include [some CSV files](https://github.com/arkypita/LaserGRBL/tree/master/LaserGRBL/CSV)  that contains "translations" from numerical error codes, alarms **and settings** that allow LaserGRBL itself to print out user friendly messages instead of cryptic numbers.

LaserGRBL decrypt the numeric value from grbl messages asap it receive them, so they become part of the message as a brief description in parenthesis. LaserGRBL also decode an extended description shown as tooltip.

When exporting configuration a "background transmitter/receiver" will send $$ command and wait for any reply from grbl side, and stop when detect a small amount of silence time after last message received (grbl finish sending config). All the contents of the reply goes to the file, included the brief description.

When importing configuration the "background transmitter/receiver" send the whole content of the file in the same manner if you open a standard gcode file via "file-open" and stream in standard mode.
Grbl itself is able to strip-out all the comments, and [parenthesis are treated as comment](https://github.com/grbl/grbl/wiki/Frequently-Asked-Questions#configuring-grbl)

The only difference is that import process will check the content of the file and send only lines starting with $ so if you open a standard gcode (like an engraving job) no lines will be sent.

> This whole thing come after I got errors and the final 4~5 settings where not updated when importing a previously generated configuration. after trying several times (about 4) it all went well.
probably a longer delay between each line could help? ....

I have already experienced this problem with my hardware. I don't know what is related about. My LaserGRBL code is OK from its side, maybe grbl cannot handle multiple $= commands without time between them? I have read something about that somewhere

like here: https://github.com/gnea/grbl/issues/165
here: https://discuss.inventables.com/t/modify-multiple-grbl-settings/39080/12

finally [Known bugs](https://github.com/grbl/grbl/wiki/Known-Bugs#character-counting-streaming-protocol-issues)

> When Grbl stores data to EEPROM, the AVR requires all interrupts to be disabled during this write process, including the serial RX ISR. This means that if a g-code or Grbl $ command writes to EEPROM, the data sent during the write may be lost. This is usually rare and typically occurs when streaming a G10 command inappropriately inside a program. For robustness, GUIs should track and detect these EEPROM write commands and handle them appropriately by waiting for the queue to finish executing before sending more data. Note that the simple send-response protocol doesn't not suffer from this issue.

Ok... simple solution! Just use send-response for the "background transmitter/receiver" while importing config.

arkypita, 2017-08-31:
> This whole thing come after I got errors and the final 4~5 settings where not updated when importing a previously generated configuration. after trying several times (about 4) it all went well.
probably a longer delay between each line could help? ....

should be fixed in v2.8.2
if you will test tell me if it change its behaviour

closed 2017-08-31 by arkypita

jorgemarmo, 2017-08-31:
Awesome!
You have done a great job with lasergrbl! Thank you for sharing!!

Can't wait for the implementation of vectorial graphics.

I'll test it this weekend and I'll let you know.

Best regards.