arkypita/LaserGRBL #87: Would like to know how to adjust or calibrate the steps of X and Y stepper motor on Laser GRBL. Thanks.  Mar

mgrgas26, 2017-09-04:


arkypita, 2017-09-04:
Maybe you mean "how to calibrate... **on grbl**", not on LaserGRBL.
Yes, because the settings you looking for is not in LaserGRBL but in grbl.

Grbl is the software that stay on arduino and translate g-codes like G1 X10 Y20 in motor pulses. Grbl can be configured sending simple "$param=value" commands. The params that control steps are $100, $101, $102

https://github.com/gnea/grbl/wiki/Grbl-v1.1-Configuration#100-101-and-102--xyz-stepsmm
http://www.diymachining.com/grbl-settings-101-a-how-to-guide/

A full list of all the params and their meaning could be found on [official grbl wiki](github.com/gnea/grbl/wiki/Grbl-v1.1-Configuration)

arkypita, 2017-09-04:
There are two modes to compute the correct value for $100, $101

1) use `steps_per_mm = (steps_per_revolution*microsteps)/mm_per_rev` formula (see links in prev post)
2) send G91 then send G1 X10 Y10. this command should move 1cm in each direction. Measure the real movement and if it move 2cm you should halve the current $100, $101.. value, if it moves 0.5cm you should double it. Use a simple linear proportion to compute the correct value, then send a bigger movement like X100 Y100 (move 10cm) and measure again to obtain a more accurate value

mgrgas26, 2017-09-04:
I have not seen any button or field on LaserGRBL to make such adjustment of values.  On GRBL controller 3.6.1, I can see that under advanced button.. Thanks.

arkypita, 2017-09-04:
any "$param=value" can be submitted as any g-code command using this textbox

[image]

mgrgas26, 2017-09-04:
It seems I am not getting the right LaserGRBL Software as per attached photo.
[image]
Could you please send the download link for the LaserGRBL software you are using? Thanks for your help.
Mar

arkypita, 2017-09-04:
Your version is correct but you must connect to your grbl prior to sending any command.
Have you ever tried to press the connection button near COM and Baud definition?

mgrgas26, 2017-09-04:
When connected to the machine, I did not see dollar code displays on my laserGRBL software just like yours, but I am able to operate the laser engraver.  Yup, the machine is connected on Com 5 at 115,200 Baud. The      machine is engraving. Thanks.  Mar

arkypita, 2017-09-04:
You can type, write, input, digit whatever you want in this box.

"dollar code" does not appare magically, you have to write the commands
$$ show all commands, $num=val set the config
Maybe you should read the guide I've already linked to before making any other question

https://github.com/gnea/grbl/wiki/Grbl-v1.1-Configuration

mgrgas26, 2017-09-04:
Thanks for the info. Mar

mgrgas26, 2017-09-05:
Hi,
I have already made the necessary adjustments for the X and Y steps ($100 and $101) and the machine now executes the exact movement as programmed.  Thanks.
Mar

arkypita, 2017-09-05:
well done

closed 2017-09-05 by arkypita