arkypita/LaserGRBL #89: Ayuda con modo laser

DieC265, 2017-09-06:
Hola.
Al utilizar el programa se me plantea el siguiente problema:
Teniendo en la configuración de GRBL el modo laser activado ($32=1) y en el propio programa LaserGRBL el modo laser activado a veces funciona y otras no la no detención del eje de movimiento del laser, es decir pasa encendido por la parte negra de la imagen que debe quemar y al llegar a una parte que no debe marcar el eje frena, apaga el laser, frena de nuevo cuando llega a una parte negra a marcar y enciende el laser siguiendo así siempre. 
El movimiento del cnc no es fluido porque se detiene cada vez que debe apagar y encender el laser.
Cambiando la velocidad de grabado y la velocidad de movimiento que aparece en la pestaña cuando creas el proyecto a grabar muchas veces se soluciona ese problema y hace el movimiento fluido sin detener el cnc solamente enciende y apaga el laser, pero después de varios proyectos cuando vuelvo a poner las mismas velocidades que venían funcionando bien me vuelve a pasar que se detienen los ejes en vez de solamente encender y apagar el laser.
Era para consultar si estoy configurando algo mal en el programa o qué podría ser lo que genera ese problema.

arkypita, 2017-09-07:
I'm sorry, I do not understand Spanish and automatic google translation is not very understandable. I hope someone else can help you.

Laser mode in grbl ($32=1) only provides effects if it is used with the M4 command, read about laser mode here https://github.com/gnea/grbl/wiki/Grbl-v1.1-Laser-Mode

The "laser mode" switch in LaserGRBL settings should reflect $32 state, **but only alter the way the image is rendered in preview** and has no practical effect on the result of the engraving.

brakthehun, 2017-09-07:
¿Qué velocidades tiene fijado en?
¿Cuál es su SMax establecido en?
Puede estar sobrecargando los búferes.

hmikyas, 2017-09-11:
hello i am from turkey hakan;
  I am using lasergrbl but I have a problem with program y and x axis is slowing while google crome is off accelerating again when crome application is open how can i solve this problem please help
mail: hmikyas@gmail.com

jorgemarmo, 2017-09-26:
For the other people reading, here is a condensated translation... DieC265 says that its laser axis stop, then turn on the laser, then movement is resumed when entering into an engrave zone and the opposite when leaving one, axis stop, laser goes off, axis resume movement.
He says that sometimes changing the speed solves the problem but some other times with the same 'good' parameter it doesn't solve its problem.
So he has this -----------|............|------------
Instead of this -----------.............-------------

----- axes moving laser off
..... axes moving laser on
|     axes stopped laser change state (on/off)

closed 2017-10-02 by arkypita

ingrc91, 2020-04-17:
Buen dia a todos, alguien puede ayudarme, tengo problemas con mi laser de 450mW, anteriermete me estaba funcionado de maravilla actualmente volvi a probarla despues de tiempo y ya no genera señal ttl/pwm(se mantiene apagado), el laser sin esos terminales puestos en la cnc shield prende sin problemas