arkypita/LaserGRBL #91: Engraving reverse

Yakov76, 2017-09-07:
Hi ,
Great program! I like very much, but ... upload photos, but on the contrary engraves (( Perhaps you need to load a negative photo ? Am I doing something wrong? For engraving on a stone perfectly, but on wood the negative turns out.

Best regards,
Yakov

arkypita, 2017-09-08:
Do you mean you have white where should it be black, and black where should it be white?

LaserGRBL produces grayscales images by varying the laser power through the S command. The variation occurs between a minimum S-MIN value and a maximum S-MAX value that can be set in the "[target image size and options](http://lasergrbl.com/en/usage/raster-image-import/target-image-size-and-laser-options/)" window.

The S-MIN value (usually zero) corresponds to the value that will be sent to the laser to make pure white, and corresponds to the laser off or to the lowest power.

The S-MAX value (usually 255) corresponds to the value of darkest black, corresponding to the laser at its maximum power.

If you have S-MIN - SMAX values of 0-255 your laser must be turned off at S0 and turned on at maximum power at S255. This is the convention for power modulation. If that is not the case, your laser has some problems.

Also grbl parameters 30 and 31 may affect this
https://github.com/gnea/grbl/wiki/Grbl-v1.1-Configuration#30---max-spindle-speed-rpm

Yakov76, 2017-09-08:
Yes exactly !
Settings 0 and 255 ... I will test and write the result. Thank you !

Yakov76, 2017-09-08:
The problem is canceled! I registered $ 32 = 1
Thank you so much !

closed 2017-09-08 by Yakov76