arkypita/LaserGRBL #94: Jumping around could be minimized

lnxbil, 2017-09-10:
Hi,

This is not a bug, only some optimization request:
I noticed some jumping and therefore increased runtime (e.g. the eyes):

[image]

I try to prevent this manually by running smaller/more local jobs. Maybe there is some way to optimize this.

arkypita, 2017-09-11:
Path optimization [was already discussed](https://github.com/arkypita/LaserGRBL/issues/82) and added to [development roadmap](https://github.com/arkypita/LaserGRBL/issues/64).

However it is a tedious discourse, establishing a better path is not as easy as it seems.
Also conflicts with needs of who use the [laser to cut materials](https://github.com/arkypita/LaserGRBL/issues/18).

closed 2017-09-11 by arkypita

Glasairman, 2024-07-24:
This is an Inkscape/JTech extension generated gcode file. It takes forever, does 2mm here , 2mm there  and jumps around generally like a fruit fly on ecstasy. 
Any ideas ? 
[image]