arkypita/LaserGRBL #95: Include a Z move when using multiple passes

Grover49, 2017-09-11:
I was wondering if it would be a simple process to add a Z modifier box by the  Perform multiple passes box.

arkypita, 2017-09-11:
Not so difficult, but not n°1 priority for me now. I'll add to development roadmap but cannot say when and if i will implement it (currently I am rewriting the whole g-code generator to be more maintainable).

closed 2017-09-11 by arkypita

Grover49, 2017-09-11:
I completely understand, I certainly appreciate what you are doing and want to sincerely thank you for what you have done with LaserGRBL.

mschaeck, 2019-01-05:
This would be a good feature to laser bigger parts

arkypita, 2019-11-23:
You can test this version: https://github.com/arkypita/LaserGRBL/releases/tag/v3.0.19
Maybe custom gcode between passes can be a solution for you.