arkypita/LaserGRBL #96: Speed issue with google chrome

arkypita, 2017-09-12:
hello i am from turkey hakan;
I am using lasergrbl but I have a problem with program y and x axis is slowing while google crome is off accelerating again when crome application is open how can i solve this problem please help.

arkypita, 2017-09-12:
I had already experienced some problem with chrome, but this time it sound very strange since laser speed is controlled directly by arduino and cannot be influenced once gcode was sent to it.

Please provide this information: windows version, chrome version, kind of hardware (laptop or desktop, how many ram, cpu model). Also try to distinguish if the slow-down of laser is a real slow-down (laser is really going slower) or it is a hiccup (discontinuity) in sending gcode.

arkypita, 2017-09-12:
Here is what i found (already seen in my job experience)
I use a streaming thread with a forced task switching for each cycle (asking for 1ms delay). Windows usually cannot do this small timing, because it use a 15ms timer for multitasking, so I have a 15ms stop each command I send.

Sometime chrome can reprogram system timer to use 1ms instead of 15ms, so windows can accomplish my request of 1ms delay and my thread goes faster.

I am going to fix/improve this beahviour, thanks for your reporting

hmikyas, 2017-09-12:
hello thanks for the answer;
my pc laptop i5 cpu 8gb ram operating system windows7 ultimate service 1

hmikyas, 2017-09-12:
y and x are too slow when my machine is off chrome crome is accelerating when I open your fix

hmikyas, 2017-09-12:
you are truly a perfect person i love you very much you are very good greetings from turkey

arkypita, 2017-09-12:
please check this pre-release and tell me if you have changes
https://github.com/arkypita/LaserGRBL/releases/tag/v2.8.4

hmikyas, 2017-09-13:
yes it was super this new version is good no problem no longer fixes worked

hmikyas, 2017-09-13:
v2.8.4 versiyonda sorun çözüldü çok teşekkür ederim :-)

closed 2017-09-14 by arkypita

hmikyas, 2017-09-26:
Thank you

iPhone'umdan gönderildi