arkypita/LaserGRBL #98: How to upload GBRL HEX file to Arduino?

CJsolar, 2017-09-13:
I am new to Arduino.  I need to upload the grbl_v1.1f hex file into the Arduino so I can run the Laser GRBL program.  Can some one tell me how you do it.  I tried XLoader but when I try to connect with Laser GRBL program it just says connecting... but never does.  I must be doing something wrong.

arkypita, 2017-09-14:
LaserGRBL is only a frontend for grbl. This king of requests should be placed on the grbl page, not here.
https://github.com/gnea/grbl/wiki/Flashing-Grbl-to-an-Arduino

Also google is easy to use: https://www.google.it/search?q=XLoader+grbl

arkypita, 2017-09-14:
If your arduino is a cheap chinese clone with CH340G serial adapter, please be sure to have the proper drivers installed in your OS. Google for “arduino CH340G” if your system does not recognize serial port.

[image]

closed 2017-09-14 by arkypita