arkypita/LaserGRBL #100: Z Axis Movement.

StuartB4, 2017-09-14:
Just wondered if it would be possible to add a feature that would allow the up and down movement of the Z Axiz (Laser up and down) to allow for focusing for different thickness's of material.
Or possibly an option to switch the laser on at low power to project a laser dot to allow for manual focusing of the laser lens before burning.

arkypita, 2017-09-14:
> Just wondered if it would be possible to add a feature that would allow the up and down movement of the Z Axiz (Laser up and down) to allow for focusing for different thickness's of material.

[custom button](http://lasergrbl.com/it/usage/custom-buttons/)

> Or possibly an option to switch the laser on at low power to project a laser dot to allow for manual focusing of the laser lens before burning.

[custom button](http://lasergrbl.com/it/usage/custom-buttons/)

closed 2017-09-14 by arkypita

StuartB4, 2017-09-14:
Thanks very much Arkypita. I forgot all about the Custom options.

StuartB4, 2017-09-15:
Thanks again. I had a go and did it. I've told it to only move in 0.25mm steps because it's for focusing and it seems to work ok.
I borrowed the Up and Down icons from your icons in the Grafica folder in the source code and made them slightly smaller so I could get the letterZ in there as well.
[image]

arkypita, 2017-09-15:
very good!