arkypita/LaserGRBL #102: acceleration offset

brakthehun, 2017-09-14:
I know I've asked for a lot. I'm so sorry,, but I've come to standardize on this software for almost anything I do.
Anyways...
I notice when My picture is a light grey at the edges,, it leaves a darker mark there. I assume it's because M4 isn't perfect. It would be nice to have an acceleration offset you could set depending on the acceleration of the machine. So,, I would have a blank 5mm or so that the laser would go to.it would start accelerating to laser speed with in that 5mm offset,, then start firing the laser once it hits the picture edge,, fire it alll the way to the next edge,, then have another 5mm to decelerate and then re-accelerate back the other way for the next line.

arkypita, 2017-09-15:
Yes, i know the problem. I also think that is because of M4 that rise-up the power too much when strong-decelerate. Did you just report this on grbl github?

Could be a good feature, i am planning to rewrite the whole gcode generator of lasergrbl, i think could be easy to add when i finish this task.

closed 2017-09-15 by arkypita

brakthehun, 2017-09-15:
With that grayscale graph from the speed modulation enhancement,, I don't think M4 could ever be accurate for all lasers on all materials at all accelerations and deceleration. I assume M4 is linear.