arkypita/LaserGRBL #99: s-min has no effect?

askmenow, 2017-09-14:
In target-image windows, set s-min to any non-zero eg. 100
 ($32=1, M4 or M3)
in the output nc file,  u can see S0 , not S100

arkypita, 2017-09-14:
can you upload a sample of gcode? (file-save program, then dragdrop the file here)

arkypita, 2017-09-14:
please note that Smin-Smax is used only to modulate the image pixels, but also laser move to go to next line without tracing. the line advancement movement is done with S0 independently from Smin value, so it is normal to see some S0 in file.

askmenow, 2017-09-14:
-------------------------------------
M4 S0
G1 F100
G91
Y38 S1000
S0           <----------------1
G0 X0.125 Y0 F4000 S0
G1 F100 Y-38 S1000
S0
G0 X0.125 Y0 F4000 S0
G1 F100 Y38 S1000
S0          <----------------2
G0 X0.125 Y0 F4000 S0
G1 F100 Y-38 S1000
S0         <----------------3
...
---------------------------------
i set s-min=100, ouput 1,2,3 is S0, how to use s-min?

closed 2017-09-14 by askmenow