arkypita/LaserGRBL #103: Posizione immagine e punto partenza

caparrottapasqualino, 2017-09-15:
Buongiorno, complimentandomi per l'ottimo software vorrei porre due quesiti:
1. se importo un G-code con posizione iniziale centrale non riesco a visualizzare sul vs software tutta l'immagine ma soltanto la porzione in alto a dx senza possibilità di zoom/dezoom. Esiste possibilità di vedere l'immagine da incidere nel suo complesso?
2. Esiste la possibilità di effettuare una sorta di anteprima di incisione per vedere l'area di incisione impegnata o comunque capire qual è il punto di partenza dell'incisione?

Grazie in anticipo, saluti.

Pasqualino

arkypita, 2017-09-15:
Ciao

1. No, per come è scritto il codice di LaserGRBL la visualizzazione è centrata su 0,0 e taglia le parti negative del piano. Ho già inserito nella roadmap #64 l'obiettivo di rendere più virtuale la routine di disegno per visualizzare anche il negativo.

2. Si, con un custom button è possibile. Qui trovi un esempio di come farlo.
http://lasergrbl.com/it/usage/custom-buttons/

arkypita, 2017-09-15:
sarebbe preferibile scrivere in inglese

closed 2017-09-15 by arkypita

caparrottapasqualino, 2017-09-15:
Perfetto. Grazie mille per la risposta.

Chiedo scusa per l'italiano, eventuale prossimo post scriverò in inglese.

Saluti