arkypita/LaserGRBL #104: setting home position

yanard3, 2017-09-15:
every time i try to set the home position it just gives me a error. $H ERROR: SETTING DISABLED 
i have been using EleksCAM with with my A3 5.5W engraver and was trying your software because I have some issues there but it works. what do i need to do to use your software on this cutter?

arkypita, 2017-09-15:
$H does not set home position, $H perform homing cycle
https://github.com/gnea/grbl/wiki/Set-up-the-Homing-Cycle

yanard3, 2017-09-15:
I didn't explain myself well sorry.... everything will work using Elekslaser desktop cam v3.1 including going to random position and setting it as a home position in the GUI. I assumed. Your software gives me the error when it is at the home position that i just moved it to when trying to set it using the home set button. The error I get is directly from the GUI after pressing the what looks like a set button so it isn't setting it and is telling me its disabled. OK so the error isn't about setting a home position its about me pressing a button that looks like it should set the home position.. Well that begs the original intention of my question.. How do you set the home position after you move the laser to that location so it remembers it? 
Thanks
Rick

yanard3, 2017-09-15:
I am making a video for you so there can be no mistake. :)

arkypita, 2017-09-15:
> using the home set button

LaserGRBL has NO "home set" button. LaserGRBL has HOMING button (look at tooltip).
Homing is the process to seek for limit switch and detect the physical limit of the workspace. Homing require to [be enabled in grbl](https://github.com/gnea/grbl/wiki/Set-up-the-Homing-Cycle#enable-home-cycle-and-setup-home-parameters) and require to have limit switch wired.

You mean "set zero" or "set working offset". You can do that using [custom buttons](http://lasergrbl.com/en/usage/custom-buttons/) and G92 code

closed 2017-09-15 by arkypita

arkypita, 2017-11-07:
[Version v2.8.12](https://github.com/arkypita/LaserGRBL/releases/tag/v2.8.12) can now show working position if different from machine position.