arkypita/LaserGRBL #105: Custom Button Image Size?

StuartB4, 2017-09-17:
Hope you don't mind me asking 2 questions here.

1:  What size in pixels should the images be for the custom buttons.
I've used 48 x 48 pixel png images. The image is in the middle of the frame when I save them, but when I apply them to the custom buttons the images are always slightly to the left of the frame. I can't center them in the frame. You can see the effect in image below.
[image]
Your images are in the middle of the frames but if I move mine over to the right any more they get cut off on the right hand edge.

2:  Will my custom button settings still be there after an update to any later versions, or will I have to redo them after every update?

Thanks.

arkypita, 2017-09-18:
1: It's a bug. In a first version I was using 42x42 for buttons, so the image selected was resized to 42x42 when loading the file. Now the button was enlarged to 48x48 but the image loading still resizing to 42x42.

I will include the fix in the next version, but you must re-select the images (48x48 is ok) to correct your visualization.

2: yes, and no. I am trying to keep the data persistence (config, last used raster import settings, colors etc) as stable as possible but sometime I broke something, and a version become incompatible with the previous. I think it was recently happened with the last release.

I am going to rewrite the setting system to split different setting groups in different files, to be more stable so in the future if I broke something like "remember last windows size and position" the other and more important data (settings and custom button) was preserved. I think I will broke again during this rewriting, but later it will become stable.

closed 2017-09-18 by arkypita

StuartB4, 2017-09-18:
Ok. Thankyou very much.

arkypita, 2017-09-19:
released

StuartB4, 2017-09-19:
Excellent. All custom buttons are now centered and look a lot better.
It also stored all my custom button settings after installing the 2.8.5 update.
Thanks very much.
[image]

commit 298bd076 (2017-09-18): fix #105