arkypita/LaserGRBL #109: Axis Direction Issue

MPlanck16, 2017-09-20:
Hello,

I'm having a problem when rasterizing an image both horizontally or vertically. For example, when I rasterize this image and save the G - Code file:

[image]

You can see if you visualize the G - Code file that the direction of the Y axis is inverted (Y negative). I don't know if it's intentional for some reason but with the vectorize option i'm not having this problem.

[image]

Thanks for your time an effort.

MPlanck16, 2017-09-21:
Hello,

Ok, it's clear now. I'm not using the CNC convention on my machine so i think i should invert the direction of my stepper in GRBL.

I want to use LaserGRBL to stream my G - Codes but my machine is driven by a raspberry pi with raspbian and LaserGRBL is not supported in Linux yet if i'm not mistaken.

Thank you very much for your attention.

arkypita, 2017-09-21:
You are right, LaserGRBL is windows only. I know that there are a lot of user with a "stand alone solution" based on raspberry. I just tried to develop a linux version, but I encountered too many difficulties and abandoned the idea.

closed 2017-09-21 by arkypita

Izakhamilton, 2019-01-20:
Hi There sit with a problem my direction has changed ,I used Benbox before and have gone to your program, but the homing button at the bottom left is giving me a headache , my machines home is top left, so now what happens is that left arrow on your programme lets my machine go right and left right , forwards is backwards and backwards is forwards please advise how i can resolve this issue
Many thanks