arkypita/LaserGRBL #112: Suggestion about envelope.

I8mAll, 2017-09-24:
I must start by saying than this creation is superb. Thank you.
I bought  Chinese 3w laser it came with Benbox (or name like that) then I moved on.  One neat thing they did, was have some useful buttons along the top. They gave:
Low level laser toggle
High level laser toggle
Describe a box at low level around where the job will be burnt.
The latter was useful for etching on small objects. It would be even better if an external envelope (outside trace) could be shown.
What do you think; can it be done?

Best wishes from a true admirer

arkypita, 2017-09-24:
Most of the things you ask is already available through custom buttons.
http://lasergrbl.com/it/usage/custom-buttons/

closed 2017-09-24 by arkypita

I8mAll, 2017-09-24:
Good gravy that is great stuff. Thank you so very much.