arkypita/LaserGRBL #113: $C

Yakov76, 2017-09-24:
It is not possible to check the g-code with the $ C command. I enter the command, it works, but it is not possible to start the verification process-the start button is not available. What am I doing wrong ?
Thank you !
Best regards,
Yakov

arkypita, 2017-09-25:
I cannot test it now, but from what I know the start button is enabled at this conditions:
1. Com opened
2. Program loaded
3. Status Idle

Suppose 1 and 2 are ok, what could you say about 3? Is status Idle or not?
Note: you can read status on the lower right corner of the window.

Yakov76, 2017-09-25:
Look at the screen, please
[image]

arkypita, 2017-09-25:
Greetings, you have found a bug. I did not handle the status "Check"
I will fix it as soon as possible

arkypita, 2017-09-25:
fixed

closed 2017-09-25 by arkypita

Yakov76, 2017-09-25:
Super !!! Thanks !
P.S. The program has some inaccuracies in the Russian translation. If there is a need, I will help translate Russian.
Kind regards,
Yakov

arkypita, 2017-09-26:
I am doing a lot of code and UI rewriting in this days. Maybe when i have finish we can re-adjust russian translation. Thanks

commit 2db1072b (2017-09-25): fix #113 and add "type gcode here" placeholder