arkypita/LaserGRBL #114: Bit of a problem with JPG images.

StuartB4, 2017-09-25:
Hi Arkypita.
I've just updated to GRBL 1.1 from 0.9 but have a problem.

I start off with the Laser in the Home X0,Y0 position.
Load a JPG image in.
Crop it to a certain size then click Next.
Then I try to change the size to 39 x 39 in the Width and Height boxes.
Then click Create.

When it's finished the image appears but the size has changed from 39 x 39 to 37 x 52

When I start the burning, the cross moves in the negative direction for a while and disapears off screen,
then comes back but does not follow the image lines.

Changing the image size once it's been cropped seems to upset the laser's X and Y position.
If I load a pre generated gcode file in LaserGRBL works absolutely fine.

I did a quick screen capture video here:
https://youtu.be/2fjgApE2dWc

arkypita, 2017-09-25:
1. as you can see at [0:16](https://youtu.be/2fjgApE2dWc?t=18s) your selection is not square, how do you think you can change the size to 39 x 39?
2. when you change height at [0.31](https://youtu.be/2fjgApE2dWc?t=29s) you dont'apply the value. You must press "enter" to finish the input and LaserGRBL recalculate width based on height you just change, preserving the original W/H ratio. If you click "Create!" button without press "enter" to accept the height change, the calculation will be done but you don't see the new value because the window close.
3. Image preview is based on the real content of the generated GCode analyzing the range of tracing X and Y. If your image (or your crop) has whitespaces around the object they don't result in any GCode trace, so they are not computed. Try cropping nearest to the image.

arkypita, 2017-09-25:
#36

StuartB4, 2017-09-25:
1. Now I understand why the image size changed when clicking the Create button. I thought both W and H settings were applied when clicking Create. I understand It's trying to keep the aspect ratio of the image correct, but what if I wanted to stretch the image in either direction, it won't allow me to do it. 

2. Thanks for directing me to #36 that was very helpful.

3. Just a thought about the Cropping window.
Sometimes when cropping it's hard to tell exactly where to place the first point before drawing the box.
Would it be possible to add a horizontal and vertical line/ruler that move with the mouse pointer to make it easier to get the cropping box as close to the edge of the image as possible.

Thanks very much for all your help.

arkypita, 2017-09-26:
1. I know, but who really need to stretch (which means distort) the image ratio? However, preprocessing of the image can be done with any external graphics. LaserGRBL include a minimal set of graphic editing but develop a true, well done, user friendly image editing require a lot of job and I prefer to keep the sw as much simple as possible.

3. Yes Cropping is not so user friendly, maybe when I have time i will think a better solution.

StuartB4, 2017-09-26:
Ok, thanks for your time.

closed 2017-09-26 by arkypita