arkypita/LaserGRBL #115: cnc laser grbl issue.

TerriButler, 2017-09-26:
I have the linksprite 3 axis mini engraver. I just ordered a 2.5 w laser for it and it seems to be cutting ok.  However, as it jogs around it does not turn the laser off as it changing between vectors and as it picks up and starts again in a different location.   This leaves a bunch of needless lines.  My question.  How do I get the laser to turn off as it moves around?  I am new at laser CNC (it shows eh?) and I may not be using the right nomenclature.  I hope that I am asking the question correctly.  I am using CamBam to create G Code and GRBL controller 3.6.1 to drive the machine.  Is the problem in one or both of the software programs?  Any attempt at answering my question would be helpful.

arkypita, 2017-09-26:
This is LaserGRBL software support board and issue tracker. This is not a generic "I have any problem with any hardware with any software doing something". Please address your questions to a more correct forum/group on web. Google for linksprite cnc forum, or CamBam forum, or GRBL controller forum.

closed 2017-09-26 by arkypita

TerriButler, 2017-09-26:
Arkypita with all respect I have no ideal what you are talking about.  I said I am new to CNC so I don't know any other way to ask the question.  What should I do to properly ask in order to abide by the rules of the forum?