arkypita/LaserGRBL #116: Best raster format / guidelines

jorgemarmo, 2017-09-26:
Hi arkypita and lasergrbl community,
I'm new to laser engraving, so the answer to this question might be obvious to many of you, if that's the case I apologise.....
Anyhow, I wonder what would be the best raster format to use ad an input to laserGRBL if I'm generating it from a vectorial software as illustrator or inkscape. Bmp? Jpg? Png? Compression level? Subsampling? Transparent background? Size respect to the engraved size? Pixel density?

Thank you in advance!

arkypita, 2017-09-26:
This is a very nice question! Before answering, it's important to explain how LaserGRBL works.

When you need to engrave an image of 50x50mm at 10 lines/mm you need to produce a pixel map of 500x500px to be engraved. Please note that 10 lines/mm is equivalent to a resolution of 254dpi (1mm x 2.54mm/inch) which is already a good quality.

So when you ask to produce 50x50mm @ 10 lines/mm LaserGRBL will scale your image to be 500x500px independently from the original size of the image. If original image is bigger it will be down-sampled to this size, if smaller it will be up-scaled.

The algorithm used could be selected between NearestNeighbor and HighQualityBicubic.
http://lasergrbl.com/ru/usage/raster-image-import/import-parameters/

Obviously any kind of scaling can cause artefacts, smoothing and lose of sharpness; however if the starting image has a higher resolution the down-sampling artefacts are very negligible. 
On the contrary if the initial resolution is very low the artefacts could be noticeable.

If you want to preserve the original image without any artefacts you must produce an image with a a pixel size matching exactly to desired line/mm and size.

`pxWidth = mmWidth * line/mm`
`pxHeight = mmHeight * line/mm`

jorgemarmo, 2017-09-26:
Great!
What about "border recognition" I'm leaned to think that a png (uncompressed, without smoothing) should work better. But 1) I'm not 100% sure of that and 2) black over white background? Or any color over transparent? (In the other hand transparent might generate problems is algorithm looks for "contrast"

arkypita, 2017-09-26:
About the file format and compression please keep in mind that a "pure white" pixel (RGB 0,0,0) is a special case for LaserGRBL because it allow to skip the engraving of this particular pixel/areas and use G0 fast positioning at higher speed.

The Jpg format often "stain" the image invisibly, using similar color that look the same at human eye but different for a computer. This is especially annoying with some images downloaded from the web, where the background should be perfectly white but it's not because of high compression used.

LaserGRBL can help fighting this problem with "[white clip](https://github.com/arkypita/LaserGRBL/releases?after=v2.6.9)" control, that set a threshold of what could be read as "pure white" by the conversion engine.

Jpg with high compression should be avoided, but if compression is not to high it's ok.

arkypita, 2017-09-26:
Vectorization use PoTrace algorithm. I don't know it's internal detail, so i cannot answer about your last question. If you want to know more, you can read [PoTrace tecnical paper](http://potrace.sourceforge.net/potrace.pdf).

arkypita, 2017-09-26:
Transparent background is not needed 'cause I take the image and I put it over a white background prior to any computation.

arkypita, 2017-09-26:
P.S: if you use inkscape there is a plugin to generate GCode directly from vector SVG.
Result is better then if you generate raster and use LaserGRBL to convert.

https://jtechphotonics.com/?page_id=2012

jorgemarmo, 2017-09-26:
Great, thanks!

closed 2017-09-26 by jorgemarmo