arkypita/LaserGRBL #118: JPG DISTORTION

HotelPappaGolf, 2017-09-26:
Hi  Arkypita

I tried to use your program today to engrave a jpg photo. 

The photo turned out great except that the photo as it is slanting to the left. 

I am attaching the original photo, the preview screenshot and the finished product .

[image]

[image]

[image]

Lastly it was the first time I used your program and it surely one of the easiest and best reslult producing software I have encountered in a long while. Keep up the good work

arkypita, 2017-09-26:
Thank you.

Look like an hardware problem. I am quite sure.

Could be caused by too fast acceleration-deceleration at border causing losing step by belt slippering or motor which does not hold the weight of the carriage and perform unwanted steps.
Try lower speed and lower acceleration control ([$120, $121..](https://github.com/gnea/grbl/wiki/Grbl-v1.1-Configuration#120-121-122--xyz-acceleration-mmsec2))

or

Electrical noise that cause bad interpretation of gcode
#81 
#24

StuartB4, 2017-09-26:
Could also be that the work piece was'nt held down properly and moved due to vibration of the machine.

Icemanhpg, 2017-09-29:
I will try to lower accelarepation.

Stuart B4: workpiece was held down with double sided tape. No movement was possible

StuartB4, 2017-09-29:
Oh Ok. Just a thought.

Icemanhpg, 2017-09-29:
I lowered the accelaration values and that fixed the problem.

Tganks for pointing me in the right direction

MarcoMade68, 2017-09-30:
on my CNC I solved the problem of losing the steps, lowering acceleration and fast moving speed. but in my case motors use 23 rather heavy and old drivers do not have all the current to drive at 100% i 2.8A but only 1200ma pololu48xx, waiting for dvr 8825. regards Marco

closed 2017-10-01 by arkypita

fasteddyD, 2017-11-09:
Hey Icemanhpg..  can you describe how you lowered the speed and maybe show the results of your re-burn?  You are correct, this is good software but the drifting kills the process.  I also noticed that the newest version of the software 2.8.12 has an option to slow speeds so I plan to try this shortly.  My photos currently look like your example but it drifts in both directions (left and right)

arkypita, 2017-11-09:
The option you mention does not slow the engraving speed, it slow down the communication between lasergrbl and grbl.

To slow down engraving you must act on $110, 111, 112 and $120,121,122 of grbl configuration.
https://github.com/gnea/grbl/wiki/Grbl-v1.1-Configuration

arkypita, 2017-11-09:
Note: drifting is not a software problem, its a problem of your hardware or your configuration.

fasteddyD, 2017-11-09:
Thanks arkypita... I will study and give this a try.  Although I realize many things can cause drifting, I am pretty sure it is not the laser itself as I do not have this issue when I use the software that cam with the system. With Laser GRBL I cannot double burn (runt the job a second time) because the unit changes start point on its own and I get a double, slightly overlapped image.  With the other software I can run the program several times and it does exactly the same thing each time.  My problem is a little different than the one on this post in that it moves both left and right, so on a large image it can move left and right 3 to 4 times.  It seems like something needs to go slower.

Icemanhpg, 2017-11-09:
Fasteddy. Like Arkypta said. Lower the values as indicated above. Try with values such as acceleration of 1500mm for both x and y. If it looks good increase acceleration until you notice the " drifting". That will be your limit your hardware can manage.