arkypita/LaserGRBL #119: Weak laser power option

JBSchueler, 2017-09-28:
Hi,

It would be nice if there is a weak power option available to be able focus the laser.
The amount of weak laser power should be adjustable. I noticed I need a different PWM value for the 2.5W laser as for the 300mW laser I use.

StuartB4, 2017-09-28:
Have you tried a Custom button with the M3 command to switch it on and another one for M5 to switch it off.
Not sure if it would work, but if you set the On button to M3 and then the power level to what you want it might.
For example M3 S75

JBSchueler, 2017-09-28:
Thats how I do it now, but it would be nice to have an button for that.

StuartB4, 2017-09-28:
Oh I see what you mean, a button built in to the program so you don't have to make a custom one.

JBSchueler, 2017-09-28:
Ah... sorry. Maybe I didn't explained it to well... my fault.

And when disabling the other way around by the way...
```
M5
G00 S0
```

Is it an option?

MarcoMade68, 2017-09-30:
hello to everyone, I solved this problem in this way
create a new button and paste these commands between parentheses to turn on the laser at the lowest power ($32=0 
M3 S1)

create a new button and paste these commands between parentheses to return to M4 configuration before starting burning ($32=1
 M5)

JBSchueler, 2017-09-30:
Thanks, wasn't aware that I can add custom buttons.
Consider it done, ticket will be closed

closed 2017-09-30 by JBSchueler

StuartB4, 2017-09-30:
"Thanks, wasn't aware that I can add custom buttons."
Thats strange, when I mentioned custom buttons you said:
"Thats how I do it now, but it would be nice to have an button for that."

JBSchueler, 2017-09-30:
Uhm... was misreading your text.

What I meant was I use the user command bar to enter the instructions manually.
At that point I was not aware that I was able to add custom buttons I thought you meant to make a custom button in the program just for me.

I have managed to create 2 user buttons for the 300mW laser and will add 1 more for the 2.5W laser.

StuartB4, 2017-09-30:
Oh ok. No problem :)
I like the custom buttons, makes it more your own.