arkypita/LaserGRBL #123: Enhancement? Keeping original size of imported image

e6910, 2017-09-28:
Sometimes it is desireable to mantain the original size of the imported image. If it is possible, I've missed it.

Marino

arkypita, 2017-09-29:
Hi @e6910 Marino
What do you really mean with "original size of the imported image"? Images does not have a size, at least they do not have a "physical" dimension. Images has only a size that is the number of pixels that compose the image (1600x1200 = 1.920.000 di pixel).

If you want to translate image resolution to image size you must add a concept of "screen resolution, or printing resolution". If you print a 1600x1200 image with a resolution of 10px/mm the final image size will be 160x120mm, if you print at 20px/mm the final image size will be 80x60mm

So, images does not have a physical size. I am pretty sure about that.
Sometime the original file has a "resolution" field inside it that you can use to compute a size, but not all the file format can include this field and it is not always reliable.

[image]

Maybe you mean "I want the final image size to be calculated as `image pixel size / line*mm`"?

e6910, 2017-09-29:
I'm not sure, but every image I work with in Photoshop (I'm a photographer) has its resolution embeded (EXIF information). I think (and I can be wrong) that the information you posted is taken from the file properties, and I'm not sure if Windows reads EXIF data.

If I remember correctly from my print shop days, Lines Per Inch is not directly related to the image file, but the ocmbination of printer capabilities, ink used (and how much it bleed) and paper absortion (dot gain).. so a news paper (absorp ink and has indeed an increased dot gain) will have a smaller LPI than a high quality magazine paper.

Resolution, in the case of an image file is a parameter which is not a fixed datum... The only know fact is how many pixels (wide and high). The resolution states how many of those pixels are supposed to be put in an inch. 

What I meant was to be able to keep that ratio. If the resolution is not always present, then My suggestion makes no sense.

Keep up the good work...

closed 2017-10-02 by arkypita