arkypita/LaserGRBL #124: Sleep Command.

StuartB4, 2017-09-29:
Been trying to get the Sleep command to work ($SLP). GRBL 1.1
When I type the command in LaserGRBL it responds with: 
[MSG:Sleeping} but does not actually do anything, everything is still powered including the motors.
[image]

Then I have to click Disconnect and Connect again to do anything.
GRBL comes back with:
[image]

I click the Padlock icon to unlock it and then the Reset icon to reset GRBL.
That works fine and it comes back to normal. Just can't get it to power down the motors.
Am I missing something?

Thanks.

arkypita, 2017-10-02:
LaserGRBL is only a frontend for grbl. 
What grbl does (or does not do) when a command was sent must be asked to grbl issue tracker.

closed 2017-10-02 by arkypita

StuartB4, 2017-10-02:
Ok thanks.