arkypita/LaserGRBL #125: Some requests

helmuteke, 2017-09-30:
Add some basic functions standard like

Set new Home point
Laser control ( On/Off) but low , perfect for positioning the laser
Countdown timer when running G-code

StuartB4, 2017-09-30:
To Set New Home Position.

[image]

[image]

When you've finished click Save. 
Connect Lasergrbl, move the laser to where you want it with the 8 arrow buttons and then click on your new button to set the new Home. Move the laser again and press the little house icon and the laser should goto the new home position.

helmuteke, 2017-09-30:
Hello , 

I know you can do this with custom buttons . But it will nicer if integrated.

What do you use for laser on / off ?

StuartB4, 2017-09-30:
Ok.
I just use M3 for on and M5 for off. No power settings because I dont use PWM for the laser power.
It's either on full power or off.

dgarrigop, 2017-10-04:
I use the custom button to home, but it should be home on the screen too, and only makes home for the arduino.

arkypita, 2017-10-04:
@dgarrigop 
See #36

closed 2017-10-17 by arkypita