arkypita/LaserGRBL #128: Can't connect to board

itxitx, 2017-10-03:
Hello. i'm trying to connect to the cnc machine but what i get is loud humming on x,y, and z stepper motor and the laser lit. the machine moves slowly to X+, Y+ and Z- and the status is connecting. i'm sure i chose the right port and baud according to my machine.

I can use the machine with other program (candle grbl) to do milling work. and it works normally. 

baud rate: 19200

itxitx, 2017-10-03:
additional info: when i plugged-out the USB cable, but the power cable is still connected. it stays on like that forever, humming, moving slowly and the laser lit.

arkypita, 2017-10-03:
Click on menu "? -> Open session log" and upload here your last lines.

arkypita, 2017-10-03:
what is the version of grbl installed on your machine?

itxitx, 2017-10-03:
it's grbl 1.1f
i don't know what happens, but i can connect now. 
here's what i did: - revert to older versions (no avail). then i doing some laser engraving with some other programs. then i come back to lasergrbl, update it to 2.8.6 and all of the sudden i can connect . 

from the log i see nothing strange. when i click connect it's humming and then disconnect it was still humming (it was, but now it's ok). 

03/10/2017 20.45.31	SetStatus   	Machine status [Disconnected]
03/10/2017 20.45.42	SetStatus   	Machine status [Connecting]
03/10/2017 20.45.42	OpenCom     	Open COM1 @ 19200 baud
03/10/2017 20.45.42	SendImmediate	Send Immediate Command [24]
03/10/2017 20.45.54	CloseCom    	Close COM1 [USER]
03/10/2017 20.45.54	SetStatus   	Machine status [Disconnected]
03/10/2017 20.45.58	SetStatus   	Machine status [Connecting]
03/10/2017 20.45.58	OpenCom     	Open COM4 @ 19200 baud
03/10/2017 20.45.58	SendImmediate	Send Immediate Command [24]
03/10/2017 20.46.03	CloseCom    	Close COM4 [USER]
03/10/2017 20.46.03	SetStatus   	Machine status [Disconnected]
03/10/2017 20.47.16	Program     	------------ PROGRAM STOP ------------

here's the last log (when it's normal)
03/10/2017 23.14.43	SetStatus   	Machine status [Disconnected]
03/10/2017 23.29.23	SetStatus   	Machine status [Connecting]
03/10/2017 23.29.23	OpenCom     	Open COM4 @ 19200 baud
03/10/2017 23.29.23	SendImmediate	Send Immediate Command [24]
03/10/2017 23.29.23	SetStatus   	Machine status [Idle]

closed 2017-10-04 by arkypita