arkypita/LaserGRBL #129: Z-Axis in Laser-engraver

LaserEngraver, 2017-10-04:
Hello,
I´m new in this Forum and I Need some help regarding an Special issue in my self made, combined 3D-printer-/Laser-engraver-device.

In my laser-engraver, I have also an z-axis.
The benefit is that I can adjust it to different engraving-object-hights.
It´s like a cnc-machine with zero-Points left/down/up (x/y/z).

But......
The laser is adjusted to it´s focuse-distance.
Now I want move the z-axis down, until a switch will touch the engraving-object.
The Signal of the switch should stop then the z-axis-movement and then the correct Focus-distance is adjusted to the engraving-object.

1. Does anybody know, on which pin of the arduino I Need to connect the switch (NO-Switch)?

2. What is needed in the GRBL-File to enable a function like that?

3. Is there an Option to insert a custom button for a z-axis (like the ones for x and y)?

4. Is there an command which let´s the z-axis move down until the switch will be activated?
-> The Z-end-switch will not work, because then a failure will be send back (stoped at hard Limit) and the arduino Need to be "reseted".

Thank you very much
BR
LaserEngraver

brakthehun, 2017-10-05:
https://github.com/grbl/grbl/issues/464

LaserEngraver, 2017-10-07:
O.K. probe sequence is working, like described in above link.
Following Parameters are o.k:
N10 G38.2 Z-120 F500
N20 G10 L20 P1 Z0
N30 G91 G0 Z3
But...... if an end-Switch (in homing prozedure) will be touched, the axis stops at **once and retracts** in the other direction.
If now the PROBE-Sequence is running and the Switch will be touched, the axis Need some mm to stop (like a ramp). This is not so good, because a milling tool would be pressed into the part.
What Need to be changed in the grbl-file that in case of probing the axis will stop at once when the Switch is "activated"?
thx

closed 2017-11-07 by arkypita