arkypita/LaserGRBL #130: Invert image colors (black and white) missing in import

sjsivo, 2017-10-06:
Program is super, but missing me some functions:

1. please insert "invert color" in image import (missing now this possibility for invert black and white images during import),
2. optimizing function for import only active area choose function missing (auto select only active area not whole image is need in some cases)
3. mass import more images for engraving (import more images with same parameters for importing) and insert all to one nc file. - possible in future print with laser more layers (with same or different z heigth)

Thank for support that.
Best Regards,
Bc. Milan Sivak.

arkypita, 2017-10-17:
1. easy, ok
2. you mean auto crop/trim white image border? ok
3. nice, but not so easy and not in the philosophy of LaserGRBL (minimal and easy) but never say never

closed 2017-10-17 by arkypita

sh4wner, 2019-04-01:
Hi @arkypita, 

I implemented the autocrop by adding this two functions to ImageProcessor.cs and calling it from a button in preview. It´s seems to work well in white background images.

` public void Trim()
        {
            mOriginal = ImageTrim(mOriginal);
            ResizeRecalc();
            Refresh();
        }

        private Bitmap ImageTrim(Bitmap bmp)
        {
            int w = bmp.Width;
            int h = bmp.Height;

            Func<int, bool> allWhiteRow = row =>
            {
                for (int i = 0; i < w; ++i)
                    if (bmp.GetPixel(i, row).R != 255)
                        return false;
                return true;
            };

            Func<int, bool> allWhiteColumn = col =>
            {
                for (int i = 0; i < h; ++i)
                    if (bmp.GetPixel(col, i).R != 255)
                        return false;
                return true;
            };

            int topmost = 0;
            for (int row = 0; row < h; ++row)
            {
                if (allWhiteRow(row))
                    topmost = row;
                else break;
            }

            int bottommost = 0;
            for (int row = h - 1; row >= 0; --row)
            {
                if (allWhiteRow(row))
                    bottommost = row;
                else break;
            }

            int leftmost = 0, rightmost = 0;
            for (int col = 0; col < w; ++col)
            {
                if (allWhiteColumn(col))
                    leftmost = col;
                else
                    break;
            }

            for (int col = w - 1; col >= 0; --col)
            {
                if (allWhiteColumn(col))
                    rightmost = col;
                else
                    break;
            }

            if (rightmost == 0) rightmost = w; // As reached left
            if (bottommost == 0) bottommost = h; // As reached top.

            int croppedWidth = rightmost - leftmost;
            int croppedHeight = bottommost - topmost;

            if (croppedWidth == 0) // No border on left or right
            {
                leftmost = 0;
                croppedWidth = w;
            }

            if (croppedHeight == 0) // No border on top or bottom
            {
                topmost = 0;
                croppedHeight = h;
            }

            try
            {
                var target = new Bitmap(croppedWidth, croppedHeight);
                using (Graphics g = Graphics.FromImage(target))
                {
                    g.DrawImage(bmp,
                      new RectangleF(0, 0, croppedWidth, croppedHeight),
                      new RectangleF(leftmost, topmost, croppedWidth, croppedHeight),
                      GraphicsUnit.Pixel);
                }
                return target;
            }
            catch (Exception ex)
            {
                throw new Exception(
                  string.Format("Values are topmost={0} btm={1} left={2} right={3} croppedWidth={4} croppedHeight={5}", topmost, bottommost, leftmost, rightmost, croppedWidth, croppedHeight),
                  ex);
            }
        }`

I have it working in my local branch.