arkypita/LaserGRBL #132: laser setup for 3 axis engraving machine.

TerriButler, 2017-10-07:
Is there a way to turn the laser off as it moves over to new vectors.  I keep getting the burned lines that I do not want.  I have been able to edit the G code to enter M5 every time it jogs but that is very tedious.  There seems to me that the CAM software or controller software should turn the laser off automatically

StuartB4, 2017-10-08:
The program you use for generating the GCode should add the on / off comands in the correct places when it works out the path for the laser.

TerriButler, 2017-10-08:
Thanks  StuartB4.  I am using CamBam to generate the GCode.  It does not add the on/off.  I am not sure if CamBam has laser capability.  Do you know of another program that is better for laser?

StuartB4, 2017-10-08:
LaserGrbl will do it it'self if you just load an image in to it, and then click on the Vectorise option.
I only use LaserGrbl or Inkscape with GCode extensions added but there are a lot of programs that do it now. 
This video might help.
https://www.youtube.com/watch?v=6gvnLMa58qQ

TerriButler, 2017-10-10:
Thanks StuardB4.  I like the laserGrbl better than the grbl controller I have been using and CamBam I have been using.  Seems much easier.  Now if I can figure out how to configure Homing and finding the origin I think I will be set.

brakthehun, 2017-10-26:
When you put GRBL in to Laser mode enable ($32=1) It will automatically shut off the laser unless it has a G1 command. G0 jog commands won't fire the laser.

closed 2017-11-07 by arkypita