arkypita/LaserGRBL #133: G-Code from inscape gcode generator causes crash

wyzarddoc, 2017-10-07:
every time i attempt to load a gcode file generated by jtech gcode generator for inkscape laserGrbl crashes. Seems the 2600 line file is too big for program. 

[Design1_0002.txt](https://github.com/arkypita/LaserGRBL/files/1366486/Design1_0002.txt)

StuartB4, 2017-10-08:
Same here. I downloaded your file, loaded it in to LaserGrbl and it crashed without even displaying anything.

arkypita, 2017-10-08:
I usually use jtech gcode generator plugin without problems.
LaserGRBL has no limits on huge file size, and can handle more then 2 million lines program (tested)

However has problem with negative space, because it's math presupposes that the drawing is all in the positive quadrant of the Cartesian plane. Your drawing seem to be all in the lower right quadrant, with positive X and negative Y.

In some cases he succeeds in drawing a piece of the work, but in other cases (as in this) the number send him into crisis.

arkypita, 2017-10-08:
A fix is already in development roadmap.

closed 2017-10-08 by arkypita

arkypita, 2017-10-08:
try last release. your file look like this.
check size and position... maybe your file is wrong

[image]

commit ab1eabb3 (2017-10-08): fix #133 and implement #66

#133
#66