arkypita/LaserGRBL #138: Just ”connecting” not connect

rostorp, 2017-10-12:
I will use it för my BENBOX mini laser engraver 300 mW. I am running Win XP, when I connect the USB it show me PORT 3 (same as in BENBOX software) when I press ”connect” it give me status ”connecting” then happens not more.
I have also tested with Win 10 computer, with the same result. The BENBOX software work, but i do not like it.
Any suggestions what to do ?

arkypita, 2017-10-12:
I think benbox use its own firmware.
To use your engraver with LaserGRBL you must install grbl on your engraver.

Not so easy and not so difficult, depends of your skills on flashing arduino.
Also you need to reprogram/reconfigure your system

here a starting point: https://benboxlaser.us/index.php?topic=11.0
please note that the document is old, now the last grbl version (1.1) is on https://github.com/gnea/grbl

rostorp, 2017-10-12:
Ok, so the controller card is an Arduino, if I flash, then I have left the chans to use BENBOX software ? I have never flashed an Arduino, but some other eprom, I think I can manade it.

arkypita, 2017-10-12:
Yes, you are right. If you flash grbl you cannot use benbox sw until you restore benbox firmware, but you can use a lot of better sw!

(restore benbox should be easy http://www.instructables.com/id/Benbox-With-Eleks-Laser-Engraver-GearBest-Banggood/ step 5)

arkypita, 2017-10-12:
maybe someone can help you here or in benbox forum http://benboxlaser.us

arkypita, 2017-10-12:
http://benboxlaser.us/index.php/board,13.0.html

rostorp, 2017-10-12:
Thanks, I make a try to flash, if it is no succsess, I have to start over again, one problem at a time.
Thanks for your help so far

closed 2017-10-14 by arkypita