arkypita/LaserGRBL #139: Max Size on XY Axis Limited

eganfrantz, 2017-10-16:
First and foremost, let me congratulate you; this is by far the best open source laser software I have found. Thank you! 

See below my 9 x 9 foot laser engraver I'm now operating with LaserGRBL and Arduino Nano. This is the 2.5w prototype. Now that everything appears to be accurate... when this engraving completes (5 days and running), I'm swapping out the laser for a 15w.

[image]

The only issue I am experiencing is max print size. It seems that once I go over 1000mm or so the little input boxes go red, as though I'm exceeding some built in limit. Is there a way to unlock this limit in next update?

Thanks again!
Best,
Egan

arkypita, 2017-10-17:
Woha! This frame is awesome :-)

The red border is only a safety hint, not really limiting the input. I think you have already tried with numbers > 1000 and it work, right?

However, there is an hard limit of the maximum size and resolution due a limit of the Bitmap object and of application memory limit that in 32bit environment is near 2GB per process.

If you need to work with 1000x1000mm at 20line/mm LaserGRBL need to internally produce a 20'000*20'000 pixels bitmap (whatever it is the original image size). Each pixel needs 32bit, so an image of this size require 20000*20000*32/8 = 1.6GB

Because the whole process require some steps, and each step require to work on a bitmap, sometime I can have 2-3 of this size bitmap in memory, so the 2GB limit is easly reached and the program crash.

I have experimentally found that I can manage max 6000*7000 that means a size of 300x350 @ 20lines/mm (or 600*700 @ 10lines/mm  or 1200*1400 @ 5lines/mm etc). I have hardcoded this limit on LaserGRBL so the line/mm resolution is internally limited to a reliable value (despite what set as request).

So... yes, you can use a bigger image size, up to any size you want, but LaserGRBL will limit the resolution.

arkypita, 2017-10-17:
But you are lucky!

First: modern 64bit OS does not have the limit of 2GB per process, so I can easly remove my hard-coded resolution limiter to allow bigger image processing on 64bit environment (i will do that just for you in the next release).

Please take note that if I do that, the processing of a big image of 9x9 feet @ 10 line/mm require 
27'430x27'430 pixels bitmap to process (near 3GB of data!). Due to the needs of LaserGRBL to use more than one bitmap in memory 3GB could be 9GB during the processing.

If your pc does not have such quantity of memory the whole process could be very very slow, because the needs of swap on disk (Windows use slow hard drive as memory when you don't have enough physical memory).

Second, there is a lot that I can optimize in LaserGRBL to make it capable of handling this huge image size:
- remove the needs to use more than one image in memory (require less memory)
- some optimization in gcode generation (speed up the whole process)

eganfrantz, 2017-10-17:
Yes, in fact, you are correct, they are just hints... 
You are a God.
Looking forward to the updates.
And thank you for your quick and detailed reply!
Best,
Egan

arkypita, 2017-11-23:
Hi @eganfrantz I have removed the 6000x7000 resolution limit.

Now LaserGRBL can handle up to 22000*22000 (480Mpx) that mean an image of 2.2mt x 2.2mt (7x7feet) @ 10lines/mm of resolution.

I have tested it with 1mt x 1mt @ 10lines/mm (100Mpx) and it take about 10 minutes to generate the GCode on a 2.8GHz quad-core - using 5GB of RAM. It may be a little slow, but i have margin for improvement working on further optimizations.

For now it's all, you can try these changes by downloading [pre-release 2.8.21](https://github.com/arkypita/LaserGRBL/releases/tag/v2.8.21)

arkypita, 2017-11-23:
Let me know if you will do some test with your incredible equipment :-)

closed 2017-11-23 by arkypita