arkypita/LaserGRBL #143: Cutting frame in engraved pictures

LaserEngraver, 2017-10-20:
Hey,
today I go another crazy idea.
The Software is nearly perfect for Picture engraving. 
If you want to insert a picture, engraved in a thin pcs of Wood (inlay) into a Picture-Frame, it would be fine to create a lasered Frame around the Picture as "cut-markers".
Means, the Software ask if you want to create a cut-frame / cut-markers around your Picture.
If yes, you can choose the size of it and it´s Position to the engraving-picture.

BR

brakthehun, 2017-11-03:
I think you can create a button to do that.

arkypita, 2017-11-07:
I understand your suggestion, but it was in the direction to make a tool for "add path, text, lines" already asked #85

closed 2017-11-07 by arkypita