arkypita/LaserGRBL #145: can't make it work.

juandrf, 2017-10-22:
Software looks to be the one I need but, after I installed and run it, it only performs a few movements and then it stops while the laser remains on. I have to reset to make it work again but again, only 2 seconds after job starts,then it stops. Any comment?

arkypita, 2017-10-23:
I am not a magician, please add more info:
- how exactly stop? can you take a screenshoot of LaserGRBL when the job stop?
- what version of grbl are you using? (the number that appare when you connect)

[image]

juandrf, 2017-10-23:
Thanks for your quick response.

Once I open the program, i can hit exacly 6 times any of the arrow keys and then the program is dead. I just downloaded ver. 2.8.8

I also did try several baud speeds and only with 115200 could see some action.

Im using a toshiba lap top with windows 10. Im also attaching a screen shot but no error is shown.

The engraver is a 2500mw from china, received unassembled. It worked with its original software, then I used another one (ELEKSMAKER) and both work good, but they cant do the outline at the end of the printing. Actually, they do it but they do it one little dot at the time. And I need this feature for cutting vinyl or paper.

Thanks and have a great day!

[image]<https://user-images.githubusercontent.com/8782035/31877938-a454ce6e-b7d8-11e7-859c-b3accee2a74c.png>

[https://user-images.githubusercontent.com/8782035/31877938-a454ce6e-b7d8-11e7-859c-b3accee2a74c.png]

arkypita, 2017-10-24:
Please answer to the second question:
- what version of grbl are you using? (the number that appare when you connect)

juandrf, 2017-10-24:
2.8.8

StuartB4, 2017-10-24:
Juandrf. 
2.8.8 is the LaserGRBL version number.
He wants the GRBL version number.
If you switch your machine on and click Connect it should show you the number like in the picture he posted.
It will be something like: Grbl 0.9j or Grbl 1.1 or earlier . (The GREEN text in the left hand pane)

juandrf, 2017-10-24:
version 0.8

Thanks!

[cid:aaaaf384-1477-402d-957e-ef06d5fa91eb]

arkypita, 2017-10-25:
As you can read on the home page LaserGRBL is compatible with Grbl v0.9 and Grbl v1.1 because v0.8 is a very old version and I never tested it before. However, I have installed grbl v0.8 on my arduino to check if it works, and it looks working.

I cannot guess what could be the problem, i suggest you to take a screenshot of the situation when it happens, and copy paste here with the session log content (menu ? - open session log)

I suggest to update to v1.1 that has a very better implementation for laser

juandrf, 2017-10-26:
How can I get the newer version?

Can you tell me the link or web site where I can download it?

Again, I can provide the screenshot later today when I get home.

Thanks a lot!

arkypita, 2017-10-26:
The repository for grbl v1.1 is https://github.com/gnea/grbl
You can download the latest version from https://github.com/gnea/grbl/releases/latest

If you download the pre-compiled .hex file you can use the program XLoader to flash your arduino.
Instruction here: https://github.com/gnea/grbl/wiki/Flashing-Grbl-to-an-Arduino#for-windows

PLEASE NOTE:
- Upgrading to v1.1 will loose all settings, so take note of all configuration of your grbl. You can read configuration sending "$$" command. After update you must reconfigure your grbl by assign the correct values to each parameter.
- There isn't a one to one match between [v0.8](https://github.com/grbl/grbl/wiki/Configuring-Grbl-v0.8) and [v1.1](https://github.com/gnea/grbl/wiki/Grbl-v1.1-Configuration) config params, so you must know what they mean and associate them correctly. 
- If your previous firmware has some customization (i.e. it was compiled with different pin hardware assignment) you may lost them, so do it at your risk.

juandrf, 2017-10-26:
Thanks!!

Have a great day!

[https://avatars3.githubusercontent.com/u/3977670?s=400&v=4]<https://github.com/gnea/grbl>

GitHub - gnea/grbl: An open source, embedded, high ...<https://github.com/gnea/grbl>
github.com
grbl - An open source, embedded, high performance g-code-parser and CNC milling controller written in optimized C that will run on a straight Arduino

[https://avatars3.githubusercontent.com/u/3977670?s=400&v=4]<https://github.com/gnea/grbl/releases/latest>

Release v1.1f (2017-08-01) Release · gnea/grbl · GitHub<https://github.com/gnea/grbl/releases/latest>
github.com
grbl - An open source, embedded, high performance g-code-parser and CNC milling controller written in optimized C that will run on a straight Arduino

arkypita, 2017-10-28:
similar to #147 
did you have solved?

juandrf, 2017-10-30:
Not yet.

I haven't dedicated much time to the project.

Besides, I'm not very familiar with the upgrade procedure even with your instructions as I'm not very deep into programming. I'll see if I can get some help from one of my coworker who is in charge of the IT dept in the company I work for,

Also, the instruction book from the laser manufacturer mentions that if the nano is flashed, its program will become useless, so I'm a bit concern that I might do something wrong and then make the whole engraver machine useless or broken.

Thanks!

arkypita, 2017-10-30:
Totally agree: if you don't feel safe in flashing is better to don't do that :-)

juandrf, 2017-10-30:
I'll keep you posted on whatever comes out after I update the program.

Have a great day!

Por favor, si reenvias este o cualquier otro mail, asegúrate de usar la opción "CCO" (copia oculta) y elimina toda la lista direcciones anteriores. POR EL BIEN DE TODOS!!

brakthehun, 2017-10-31:
If it is an Ebay with a Nano controller,, you can pick up a nano off Amazon or Ebay,, flash that,, then see how it works.

arkypita, 2017-11-02:
Whould you like to do a test using an older version of LaserGRBL that doesn't have ultra fast streaming optimization ([i suggest v2.8.1](https://github.com/arkypita/LaserGRBL/releases/tag/v2.8.1))?

Let me know

juandrf, 2017-11-02:
That sounds good. I'd like to try.

I'll try the link below later today when I get home. I there any special instruction or hint to install it?

StuartB4, 2017-11-02:
I assume you have .Net Framework 3.5 installed.

arkypita, 2017-11-02:
@juandrf 
The installation of v2.8.1 will overwrite any other installation. When you run the older version you will be warned about the availability of a newer one: be care to don't update. That's all

juandrf, 2017-11-02:
I have 4.6.1 installed in my home lap top.

StuartB4, 2017-11-03:
Ok.

juandrf, 2017-11-03:
Seems like older version didn't work either.

I noticed the "send" buttom is missing but it was in the drop down menu. This older version just did samething as the newer one: moved a few steps and then it stopped while the laser is stil powered.

As I mentioned, I'm not too familiar with all these programming things, so I think I'll give up.

I really apreaciate you answering my questions may be I'm too dumb for this.

Thanks again and have a great weekend!

arkypita, 2017-11-07:
@juandrf please try [v2.8.11](https://github.com/arkypita/LaserGRBL/releases/tag/v2.8.11) and set:
- Syncronous as streaming mode
- Slow or Quiet as threading mode

[image]

Let me know

juandrf, 2017-11-07:
Thanks!

I'll try later at home.

Have a great day!

Por favor, si reenvias este o cualquier otro mail, asegúrate de usar la opción "CCO" (copia oculta) y elimina toda la lista direcciones anteriores. POR EL BIEN DE TODOS!!

[image]<https://user-images.githubusercontent.com/8782035/32420349-078de0b4-c289-11e7-9b1e-43fd82a1b10a.png>

juandrf, 2017-11-08:
My antivirus detected a threat as you can see below:

(sorry, it is in spanish).

I just want to double check before skip it. Is it safe to continue? (sorry again for the dumb question).

[cid:154dc88e-b1ef-4deb-9c6a-373ed85fc575]

Por favor, si reenvias este o cualquier otro mail, asegúrate de usar la opción "CCO" (copia oculta) y elimina toda la lista direcciones anteriores. POR EL BIEN DE TODOS!!

[image]<https://user-images.githubusercontent.com/8782035/32420349-078de0b4-c289-11e7-9b1e-43fd82a1b10a.png>

arkypita, 2017-11-08:
It's something just know #70 
Unluckily my software is sometime reported as suspicious by firewall or antivirus by heuristic detection.
I apologize for that but i cannot do anything to solve this, since its depend by false positive detection from av software.

In my analyses it is caused by the self-extractor .rar package that I use to make my installer. I confirm it is not a virus but a false positive. Some official confirm comes from [here](https://www.virusbulletin.com/virusbulletin/2005/11/false-positive-disaster-anti-virus-vs-winrar-amp-co/).

Maybe if in the future I'll change to a more professional installer.

closed 2017-11-18 by arkypita

arkypita, 2017-11-21:
@juandrf 
Whould you like to test last version? I have changed my installer and now should be more antivirus friendly.
https://github.com/arkypita/LaserGRBL/releases/tag/v2.8.20

juandrf, 2017-11-22:
Sure thing.

I'll try tonight when I get home after work.

In the worse case scenario, I'll accept a remote conection to my computer under my own responsability.

Have a great day.

https://github.com/arkypita/LaserGRBL/releases/tag/v2.8.20

juandrf, 2017-11-28:
Hi

I installed the program and still something is wrong,

And  I also tried to install Benbox and  seems like it didn't work either.

The main use that I want to give to my laser engraver is to cut thin board and foam and vinyl. I'm just learning to use Corel Draw to create the designs and I'm still looking for the way to send the drawing directly from Corel to the engraver.

Juan R

https://github.com/arkypita/LaserGRBL/releases/tag/v2.8.20

nyarsulik, 2018-01-09:
I am new to LaserGRBL and it looks like I am having the same issue.  I can get the program to start the burn, and then after about 10 seconds or so it will stop (with the laser on) and will give a prompt that "there was an error with the board" or something similar.  I'll try to find the exact wording.  Can we reopen this issue or should I try to make a new one?  

I am running GRBL 1.1.

I don't think the laser is the problem, I can turn it on and adjust it within GRBLController and it will stay on for as long as I need it to, it's only once it starts running through the GCODE that it seems to have trouble.

Thanks!
Nick

arkypita, 2018-01-09:
http://lasergrbl.com/faq/

nyarsulik, 2018-01-09:
Thanks arkypita, I've tried to change some of the settings with no success.  I'll try to swap out the USB cable and see if that might be it.  Thanks!

arkypita, 2018-01-09:
let me know