arkypita/LaserGRBL #146: progam jogging to the left half way through

ljpsr, 2017-10-25:
I have had several projects jog to the left and have an offset half way through the project. more often than not on larger sessions

brakthehun, 2017-10-31:
I had this issue when I exceeded the 30khz max pulse frequency.

ljpsr, 2017-10-31:
is that the laser itself, is there a way to change so it does not happen

brakthehun, 2017-10-31:
Let's look at your $$ settings.
How many steps per mm is yours set to?? ($100 and $101)
What is your Max speed??($110 and $111)

I had 640 steps per mm and a max speed of 10,000 mm/minute
10,000/60 = 166.6667mm/second
640 steps per mm * 166.6667 = 106666.688 pulses per second (106.7 kHz).
That was far above the 30 kHz limit.

I changed my microstepping from 1/128 to 1/32 and it worked just fine after that.

fasteddyD, 2017-11-04:
Hi brakthehun.. I am new at this so not sure how to make the change you suggest for micro stepping, from 1/28 to 1/32.  This software works better than anything I have tried for photos but this stepping right and left spoils the fun.

StuartB4, 2017-11-04:
Depends which stepper motor drivers you have, microstepping is usually done through them with dip switches or jumpers of some sort.

brakthehun, 2017-11-04:
Let's start in your command line right above the data feed. Type in $$,, press enter and screenshot the data then post it in here.

fasteddyD, 2017-11-04:
Hi brakthehun.. thanks for quick reply...this is what I get..

fasteddyD, 2017-11-04:
[screen.docx](https://github.com/arkypita/LaserGRBL/files/1442881/screen.docx)

Sorry.. cannot seem to get a screen shot to copy  above is a word file.. I will keep trying in the meantime

brakthehun, 2017-11-04:
So,, you are running an old firmware version of GRBL.
When you initially connect,, what does it say right above the "Type $ for help"

fasteddyD, 2017-11-04:
As I cannot seem to copy the print screen here, I found what you were talking about in your previous post as follows:
S100=80，000
S101=80，000
S110=3000，000
S111=3000，000

brakthehun, 2017-11-04:
That looks great. That should not be causing problems. Mine was:
$100=640.000
$101=640.000
$110=10000.000
$111=10000.000
So,, this is not your problem,, but I think your firmware might be the problem.

fasteddyD, 2017-11-04:
Hmmm... frustrating.  This software is hands down better than what came with the machine for photos but it keeps moving left and right, especially for big files, which distorts the photo.

brakthehun, 2017-11-04:
So,, when you connect,, what version of GRBL does it say you are running??

fasteddyD, 2017-11-04:
That is another problem, I cannot see where the version is located.  It does,nt show up when I load the software but I also do not have the feature where I can change speed and power settings on the fly, so I think I may have an older version

fasteddyD, 2017-11-04:
when I connect it says grbl 0.9j

brakthehun, 2017-11-04:
So,, it should be compatible with 0.9. It is better to run 1.1 though. What is your baud rate when you connect??

fasteddyD, 2017-11-04:
baud rate is 115200

fasteddyD, 2017-11-04:
I only downloaded this software about 2 weeks ago but maybe vs 1.1 came later.  Probably should see if I can download the new version.

brakthehun, 2017-11-04:
I believe 0.9 wants a baud rate of 9600
Can you try that??

fasteddyD, 2017-11-04:
ahh... yes I can....

fasteddyD, 2017-11-04:
would 1.1 use the higher baud rate?

brakthehun, 2017-11-04:
Yes.
Be careful though. The pins are for laser on and Z limit are switched in 1.1

fasteddyD, 2017-11-04:
first I will try lower baud rate with this version and hope  it does the trick.  If not, I will try downloading vs 1.1.  If I download 1.1, will I need to change anything base on your note of caution?  I am new at this and not sure what pins are or how this might create a problem.  I really appreciate you help.. I have been struggling for weeks.. trying all kinds of software, but this one is really good if only it would stop moving left and right during the burn..

brakthehun, 2017-11-04:
If you take a pic of your controller and post it,, I can help guide you.

fasteddyD, 2017-11-04:
hmm  seems it will only allow me to connect at 115200.

brakthehun, 2017-11-04:
Interesting. Dd you flash the firmware yourself??

fasteddyD, 2017-11-04:
The control board is sandwhiched between two pcs of plastic so we cannot really see it.

brakthehun, 2017-11-04:
That's alright. If you send a pic,, I could probably figure out how to use it with 1.1

brakthehun, 2017-11-04:
Where did you get the controller??

fasteddyD, 2017-11-04:
No.  I bought and assembled a kit. The board was already sandwhiched between two pcs of plastic so I just mounted it and when done, it took off without me needing to do anything else.

fasteddyD, 2017-11-04:
I will shut down and disassemble the board to take a quick photo.

brakthehun, 2017-11-04:
It might be complicated to upgrade it then. One big setback of 0.9 on Arduino is you can't dim the laser.

brakthehun, 2017-11-04:
Do you have a link to the kit??

fasteddyD, 2017-11-04:
[image]

Let me know if you can open this photo..

fasteddyD, 2017-11-04:
OH... it opened itself... hope this helps..

brakthehun, 2017-11-04:
Alright!! When you power the machine on,, does the laser cooling fan come on also??

StuartB4, 2017-11-04:
Looks like there is no option to change the microstepping on that board either.
http://www.lubandiy.com/article-3-1.html

fasteddyD, 2017-11-05:
Yes, the cooling fan comes on with power.

fasteddyD, 2017-11-05:
Is there a better controller board that I should consider?  I work in China and the supplier is located nearby.  He basically sold me the bits and I assembled them.  Mine is 6.5 watts but he did suggest a 10 watt at the time.  I think he has optional boards etc.  Any suggestion on a board that will run with this software more smoothly?

fasteddyD, 2017-11-05:
The software that came with the machine was a Chinese copy of the software supplied by Elks Maker.  It had so many problems with the English language that I finally downloaded EM's English version, which works exactly the same.. with less errors but same limitations with how it handles photos.

fasteddyD, 2017-11-05:
I tried several software package trial versions and was amazed that several would drive the laser.  This software is hands down the best for photos that I have tried, if only I could keep the laser from wandering back and forth.

fasteddyD, 2017-11-05:
I do not understand why I downloaded vs 0.9.  When I look at the available downloads I do not see an option for one or the other.  I wonder now how to determine the version of the download so that I can try 1.1??

StuartB4, 2017-11-05:
1.1 is here:
https://github.com/gnea/grbl

brakthehun, 2017-11-07:
You can use your current board to run 1.1. You just have to swap the laser ttl trigger with the z endstop.

closed 2017-11-18 by arkypita