arkypita/LaserGRBL #147: Hi friend. Please help me

RomulRS, 2017-10-26:
I have a laser engraver made of two dvd roms. When the engraver burns the picture, the outer contour of the drawing turns black and bold, and all lines inside the drawing are gray. What is the problem?
[image]

arkypita, 2017-10-26:
It seems that the original image from which you started has a very thin stroke in the inner parts.
It also seem that you have used line2line. Line2Line try to reproduce the original smoothing/gray of the source image.

Additional smoothing/gray can be generated due to scaling (ie if the original image is very big and you want to draw it small, a nice black stroke can be result in a small smooth line).

Try act on brightness/contrast, threshold or use "vectorize +filling option".

RomulRS, 2017-10-26:
Thank you for your prompt response. This happens with any image. Attached the original image.
[image]

arkypita, 2017-10-26:
I guess that if you use a DVD sled you have a max size of 40x40mm. Look like you use a resolution of about 5line/mm so the image file that LaserGRBL internally produce for engraving is about 200x200px

This is what happens when you scale a big image to a smaller one:

[image]

It look like sharp, but if you zoom it you can easly see that the smallest lines are not "pure black stroke" but they are composed of gray pixels

[image]

arkypita, 2017-10-26:
Look like that your laser doesn't have enough power to reproduce a good grey engraving, so here the cause of your poor result.

Try to use "NearestNeighbour" as scaling - instead of "HQ Bicubic" and prefer "Vectorization!" tool for this kind of images, as "LineToLine" produce better result on pictures.

RomulRS, 2017-10-26:
I saw in another program for a laser engraver, that you can choose to additionally outline the contour of the site. But there is no this setting.

arkypita, 2017-10-26:
I tell you 2 times what to do. Maybe you don't read.

RomulRS, 2017-10-26:
Thanks for your answers, I will check at home.

closed 2017-10-28 by arkypita

RomulRS, 2017-10-29:
The problem was in the high speed of the motors when burning.