arkypita/LaserGRBL #148: Cannot connect

steve-bg, 2017-10-28:
Hello arkypita,
I [purchased a kit](https://www.aliexpress.com/item/CNC-3018-laser-options-diy-mini-cnc-engraving-machine-Pcb-Milling-Machine-Wood-Carving-machine-cnc/32805289008.html?spm=a2g0s.9042311.0.0.oQ4gMu) recently, assembled it and got very exited about the opportunities it offers. The [software which came](https://www.dropbox.com/s/p7jg8ukvwft2jil/CNC%20software.rar?dl=0) with the kit works fine, but yours seems so much better.  
To my surprise I could not even connect - I clicked the button to connect and the machine started making a  weird noise (I think from the Y-axis motor, but i can't be sure) without moving and also started the laser (I think at full power).  Am I doomed with an incompatibility or is it a matter of some fine tuning(hopefully) ?  I googled for this problem but could not find anything except  #an explanation that a short weird sound at the connect time was normal. But this is not short - I waited at least 15 seconds, maybe more, then I got scared that it may damage the hardware and powered it down (closing LaserGRBL did not stop the sound and the laser). 
Please let me know if there is something I can try. I can send a photo of the board if that helps.
Thank you very much!

Dpsspower, 2017-10-28:
The laser goes on to full power because the bootloader in the control board sets all outputs high while running the bootloader routines..
After 2 seconds or so, the laser goes off again.
This has nothing to do with the software here. It is a bug in the controller board.
The controller board must run GRBL without any bootloader to avoid this
Another way to avoid the problem is switching on the laser after connecting to host.

The noise could be a function to drive all axis to the home position. If you don't have any endstop switches, the motor drives against the mechanical limit endless.
Check the GRBL settings in the board. There you can disable the homing cycle.
It is set by default after installed the latest GRBL on a board.
$22=1

Use a simple terminal program 115KBaud and connect to the controller.
Then type "$$" + Return to see the settings in the board.
If $22=1 is set type
$22=0 and return.
All settings are permanently stored in the board.
Read GRBL settings at
[https://github.com/gnea/grbl/wiki/Grbl-v1.1-Configuration](url)

steve-bg, 2017-10-28:
Thank you sir,
- As I said, after much more than  2 seconds neither the laser nor the noise stops after clicking the "Connect" button in this software.
- I don't have endstop switches, on all axes I am approximately in the middle, and as I said - nothing is moving, so definitely this is not a case of "pushing against the mechanical limit endless".
- I connected with puTTY and $22 is already 0

StuartB4, 2017-10-28:
Have you set the current for the stepper drivers. Don't assume they are set correctly at the factory.
Have a look on youtube for how to set them, unless it came with instructions with your machine.
Or have a look here:
https://www.pololu.com/blog/484/video-setting-the-current-limit-on-pololu-stepper-motor-driver-carriers

arkypita, 2017-10-28:
This is what LaserGRBL do when you click "connect" button:

- set it's status to "connecting"
- open the com port
- if success, send the ["grbl reset" (ctrl-x)](https://github.com/gnea/grbl/wiki/Grbl-v1.1-Commands#ascii-realtime-command-descriptions) command 
- start sending "?" status query and listen for status message
- as status report message contains info from grbl about the status (Idle, Alarm, Run, etc) the status change to reflect the last status received from arduino

[image]

arkypita, 2017-10-28:
You can open your puTTY and try to repeat this sequence (connect, grbl reset with ctrl-x, query status with ?) and see what of this commands cause problem to your device. I guess ctrl-x, let me know

StuartB4, 2017-10-28:
You might find some help here as well if you ask, they seem to be really helpful concerning that machine/board:
https://github.com/gnea/grbl/issues/123

steve-bg, 2017-10-28:
Thank you for the answer. This is exactly what happens with puTTY
1. The moment I establish connection a very short and weak "click" sound is heard - same happens when i start the seller provided software.  This message appears on screen:
Grbl 0.9j ['$' for help]
2. I press Ctrl-X.    Nothing moves, nothing clicks, the same message appears on screen:
Grbl 0.9j ['$' for help]
3. I type ?     Then this appears on screen:
<Idle,MPos:0.000,0.000,0.000,WPos:0.000,0.000,0.000>
4. I type $$, then this appears on screen:
$0=10 (step pulse, usec)
$1=25 (step idle delay, msec)
$2=0 (step port invert mask:00000000)
$3=5 (dir port invert mask:00000101)
$4=0 (step enable invert, bool)
$5=0 (limit pins invert, bool)
$6=0 (probe pin invert, bool)
$10=3 (status report mask:00000011)
$11=0.010 (junction deviation, mm)
$12=0.002 (arc tolerance, mm)
$13=0 (report inches, bool)4
$20=0 (soft limits, bool)
$21=0 (hard limits, bool)
$22=0 (homing cycle, bool)
$23=0 (homing dir invert mask:00000000)
$24=25.000 (homing feed, mm/min)
$25=500.000 (homing seek, mm/min)
$26=250 (homing debounce, msec)
$27=1.000 (homing pull-off, mm)
$100=800.000 (x, step/mm)
$101=800.000 (y, step/mm)
$102=800.000 (z, step/mm)
$110=5000.000 (x max rate, mm/min)
$111=5000.000 (y max rate, mm/min)
$112=500.000 (z max rate, mm/min)
$120=20.000 (x accel, mm/sec^2)
$121=20.000 (y accel, mm/sec^2)
$122=20.000 (z accel, mm/sec^2)
$130=200.000 (x max travel, mm)
$131=200.000 (y max travel, mm)
$132=200.000 (z max travel, mm)

arkypita, 2017-10-28:
The behaviour with putty is 100% ok. I can listen the click/clunch on connect also with my engraver.

But LaserGRBL does nothing different from this. Sorry I don't know how to help, but if you solve please share your solution.

arkypita, 2017-10-28:
One Idea: lasergrbl does not set the com port option, only the speed.
The other options are managed using their default from the OS. Please check your com port is configurerd for 8N1 like in image

[image]

steve-bg, 2017-10-29:
The settings are the same as yours.  
I am depressed now.  :(

steve-bg, 2017-10-29:
Even worse. I tried to generate a code by LaserGRBL and execute in the vendor's software. That caused my machine to freeze (with the laser on) just maybe 5 seconds after the start. 
[Here is the code](https://app.box.com/s/1uqe2sjpb15jzs2mkf60zwczw2kkt753) generated by LaserGRBL (looks much better but hangs)
[Here is the code](https://app.box.com/s/rgq7h5kq08dlqvii4wuhl7exr4aocl0r) generated by vendor's software (looks much worse and unoptimized but executes just fine)
Is this helping to understand what's not compatible? 
I am sorry I don't understand grbl, but one difference which is obvious is that in the coordinates the taobao code has only one significant figure after the decimal point while LaserGRBL has up to 3 significant figures.

arkypita, 2017-10-29:
It doesn't help, it only confuse. One problem a time is a better approach. However I have flashed Grbl 0.9j downloaded from official site to 3 different arduino board, configured like your configuration, and all of them work well both with lasergrbl-generated and taobao-generated file.

So:
- the code generated by LaserGRBL is ok (formally and experimentally)
- LaserGRBL streaming process itself is ok (i think)

I suggest the same hint of @StuartB4 

Ask on this thread https://github.com/gnea/grbl/issues/123 where you can find a lot of user with woodpecker controller board. They are more experienced about this controller.

itxitx, 2017-10-31:
hey. the problem is exactly the same as mine :  https://github.com/arkypita/LaserGRBL/issues/128

itxitx, 2017-10-31:
i'm using the same board (woodpecker). but i have upgraded the grbl firmware to 1.1f

StuartB4, 2017-10-31:
This is a good place to ask concerning the Woodpecker board.
https://github.com/gnea/grbl/issues/123

itxitx, 2017-10-31:
the culprit is it does happens ONLY with LaserGRBL. no other program i tried generate such weird effect.

arkypita, 2017-10-31:
I have done some changes in my code, try [v2.8.10](https://github.com/arkypita/LaserGRBL/releases/tag/v2.8.10)

itxitx, 2017-11-01:
tried the 2.8.10 , but the problem persists. plus it's marked as some kind a trojan by avast

arkypita, 2017-11-01:
I really don't know what could be the issue :-)

The false positive from AV heuristic filter is a know problem. Unlikely many AV consider suspicious any sw without digital signature.

steve-bg, 2017-11-02:
Sorry for the half-offtopic, but I could not resist to ask itxitx to clarify: 
1. The upgrade to 1.1. did not affect the use of LiteFire, is that correct? Someone stated that they had to downgrade back because they lost the ability to use LiteFire after the upgrade.
2. The upgrade to 1.1. did not help to be able to use LaserGRBL, correct?
I am sorry, I am too new in these matters, I don't even know what is a bootloader and how does the presence or absence of a bootloader affects those boards.  I am not interested to get too deep in the flashing business (not to mention soldering business).  I just want my setup to work nicely :)
Thank you all!

itxitx, 2017-11-02:
@steve-bg: 
1. i never use LiteFIre, so i don't know about this. 
2. Yup. done the upgrade and the problem persists. .

brakthehun, 2017-11-02:
I have sort of the opposite issue. LaserGRBL connects to all my machines (5 of them) and UGS is hit and miss.

itxitx, 2017-11-02:
@arkypita i'm sure the problem is related to this: https://github.com/grbl/grbl/issues/845 . the ch340G USB to Serial chip cause the problem. but i think we need to address this problem because lots of people using this, especially in my country, a lot more people using the clone arduino than the real one.

I wanna ask 1 thing maybe little off. i tried download the source code and try alter it in SharpDevelop. but when i hit Build it generate this error: Task could not find "AL.exe" using the SdkToolsPath "" or the registry key "HKEY_LOCAL_MACHINE\SOFTWARE\Microsoft\Microsoft SDKs\Windows\v8.1A\WinSDK-NetFx40Tools-x86". Make sure the SdkToolsPath is set and the tool exists in the correct processor specific location under the SdkToolsPath and that the Microsoft Windows SDK is installed (MSB3086).
i searched and tried so many solutions, but no avail. can you help with this? I never use C# or this kind of desktop project before. I'm PHP programmer.

itxitx, 2017-11-02:
@brakthehun are you using clone arduino with ch340 chip? have you tried to slowing down the settings etc?

brakthehun, 2017-11-02:
One of my machines uses the CH340 on a Nano. The rest use the 16U2 Unos. The CH340 has it's own occasional issues. I'm going to convert it to WiFi (ESP-01) but I'm having issues getting that all to work. That's another post for another time.

arkypita, 2017-11-02:
@itxitx With grbl/grbl#845 you have found a good thread, but the original problem of @steve-bg is a connection problem - not a streaming problem. However some user report me also the same issue reported on that thread (streaming start and then hang suddenly).

I say "you have found a good thread" because LaserGRBL streaming thread act in the same way the user @109JB explain.

> As further testing, I made a small program to use for testing. This program has streaming code that is very streamlined with nothing extra except what is absolutely necessary for streaming and for detecting GRBL error messages. Nothing else. Both the send-response and the character counting protocols can be used and both have generated errors.

Indeed LaserGRBL has a very fast routine for sending data, of course respecting character-counting protocol, but without adding any extra unneeded delays. This result in the ability of stream at 800 gcode line per second: https://www.youtube.com/watch?v=pK9GnzNFpHM

I have tested this with some different board, all having CH340 usb-serial chip, without noticeable problem with grbl 1.1f but I can not ignore the fact that other users have some problems.

I guess there may be two problems using this fast streaming mode:
- packet sent one close to the other decreases "silent" time on the wire and makes more statistically probable that an electrical noise will affect a data packet instead of a silence
- grbl code needs to use interrupt to serve both serial receiving and buffering, motor timing etc. i guess that serial packet can be lose/misunderstand if 

You can do a test with an arduino "off-site" or rather detached from your engraver/cnc you can exclude that it is an electrical noise problem (no motor noise).

Another test could be done using an older version of LaserGRBL that doesn't have ultra fast streaming optimization ([i suggest v2.8.1](https://github.com/arkypita/LaserGRBL/releases/tag/v2.8.1)).

arkypita, 2017-11-02:
Just for information... i have created a test file of 10'000 gcode lines all "M5" (laser off) so no movement needed and arduino execute them in zero time.

Grbl Controller takes 3 min to stream the file
SourceRabbit takes 30 seconds
Universal Gcode Sender crashes, at least on my PC
SnapCNC takes about 1 minutes and half
LaserGRBL does it job in 16 seconds

itxitx, 2017-11-02:
Guys. i found a workaround. every time it does the strange behaviour just hit the reset button and voila, it stops moving and the program connected. 
unplug the USB doesn't eliminate the problem/moving but reset the arduino does.
here's what it looks after i hit the reset button:
[image]

btw, i tried older versions up to 2.5.0 and all does the same bug. 
the bug often (almost always) resurface when i played with other CNC programs, and come back to LaserGRBL. but if i don't play with other programs the bug do not appear whatsoever

arkypita, 2017-11-02:
@itxitx thanks for sharing your workaround. 
LaserGRBL submit a reset command by itself when connecting, so a manual reset is not truly needed. 
The sequence is 
```
OpenCom();
start RX and TX thread
GrblReset(); //(ctrl-x)
QueryPosition(); //?
```

Maybe this sequence is too fast and a small delay between OpenCom and GrblReset could help.

arkypita, 2017-11-02:
Also this page capture my attention: https://electronics.stackexchange.com/questions/279629/ch340-usb-serial-interface-chip-drivers-cause-avr-reset-problems-under-windows

arkypita, 2017-11-05:
@itxitx and @steve-bg 
I have released a new pre-release [called v2.8.11](https://github.com/arkypita/LaserGRBL/releases/tag/v2.8.11)

There is a config switch to slow down LaserGRBL threads. Together with synchronous streaming mode it can solve or at least quiet down many grbl hang/issues.

[image]

Try this last version with **synchronous streaming mode** and **quiet or slow thread mode** and let me know.

closed 2017-11-18 by arkypita

arkypita, 2018-01-09:
@steve-bg 
Did you get LaserGRBL finally working with your woodpecker board or not?

itxitx, 2018-01-10:
i tried all options above, and it's the same. i have to hard reset the board sometimes.

arkypita, 2018-01-10:
@itxitx

I made a version with a debug of the communication.
You can download it here: https://www.dropbox.com/s/po8p5ee9j1zthkj/install.exe?dl=1

This version records everything going through the serial including connection and disconnection actions, in addition to the already present session log. The logs are written to the folder: C:\Users\YOURUSER\AppData\Roaming\LaserGRBL

The files in question are:
comlog.txt
sessionlog.txt

Launch LaserGRBL and press the connect button, then send me the two files.
I'll see if I can understand something :-)

Please note that comlog.txt grow fast as it record all com, so use this version only for debugging session :-)

itxitx, 2018-01-10:
sorry, i can't install it. i'm using win 10 and it seems something like the security procedure cancel the installation and delete the install file. tried few times.

arkypita, 2018-01-10:
Windows defender issue? https://github.com/arkypita/LaserGRBL/issues/235

arkypita, 2018-04-10:
v2.8.43 add hard reset via dtr-rts on connect. this could help people with issue on connecting
https://github.com/arkypita/LaserGRBL/releases/tag/v2.8.43

Configure for hard reset on connect. Let me know

mhdpop1, 2018-09-19:
Hi i have same problem with my board it is disconnecting in the middle of a job can you help me too ?

fantilator, 2018-09-19:
Try to put a powered usb hub in the middle. I had the same problem, and this is solved now.

StuartB4, 2018-09-19:
mhdpop1. If you have it running on a laptop you should make sure the charger is plugged in as well.

mhdpop1, 2018-09-19:
i am using PC and i tested everything it work 2 second and stop and not working agian

StuartB4, 2018-09-19:
Oh ok mhdpop1.

arkypita, 2018-09-19:
Hi @mhdpop1 

I can tell you about the experience of a user who had blocks and disconnections of grbl during the early stages of processing, just like you.

After many attempts and analysis we have established that the Arduino board in his possession was "defective". The biggest difficulty in understanding this problem was that with other programs like "Universal Gcode Sender" or "SourceRabbit G-Code Sender" he was able to correctly stream the gcode to the machine. Why?

This is because LaserGRBL use a very aggessive streaming thread, that is able to stream up to 800 gcode line per second, while other gcode sender are much more slower. Good quality ORIGINAL arduino boards can easily withstand the speed with which LaserGRBL communicates with them. Poor quality arduino clones can fail showing disconnection, error, hang etc.

You can solve

1) buy a original arduino board
2) set slower streming mode as suggested in lasergrbl.com/faq

mhdpop1, 2018-09-20:
> Hi @mhdpop1
> 
> I can tell you about the experience of a user who had blocks and disconnections of grbl during the early stages of processing, just like you.
> 
> After many attempts and analysis we have established that the Arduino board in his possession was "defective". The biggest difficulty in understanding this problem was that with other programs like "Universal Gcode Sender" or "SourceRabbit G-Code Sender" he was able to correctly stream the gcode to the machine. Why?
> 
> This is because LaserGRBL use a very aggessive streaming thread, that is able to stream up to 800 gcode line per second, while other gcode sender are much more slower. Good quality ORIGINAL arduino boards can easily withstand the speed with which LaserGRBL communicates with them. Poor quality arduino clones can fail showing disconnection, error, hang etc.
> 
> You can solve preventing electrical noise on wires and buying an original arduino board
> 
> As workaround with poor clones you can set slower streming mode as suggested in [lasergrbl.com/faq](http://lasergrbl.com/faq)

Thanks for reply i bought this device and it is chenesse one called DIY8000 it has CNC and laser mode
i have problem with all softwares it doesnt woek even with Gcode sender so I did install the version that you put in Dropbox for @itxitx and here i attached the files you told him to attach so can you check if you can find anything helpfull?
https://www.dropbox.com/s/avqs8v6onh00qpy/sessionlog.txt?dl=0
https://www.dropbox.com/s/9eo56n69hp2bdvk/comlog.txt?dl=0

arkypita, 2018-09-20:
[image]

As you can clearly see in comlog.txt there is a lot of weird characters in packet received by LaserGRBL.
This means that communication between LaserGRBL and Grbl on arduino has some kind of problem.

Try to do specific tests to exclude / detect cause.

1. Gently remove the arduino nano from your board (be care to not bend pins), connect to your PC via USB cable and let it talk with LaserGRBL completly detached from your CNC. Load an image and send its GCode to arduino. This test can exclude or confirm that the problem is about hardware noise from motor.

2. Try connect to another computer. This can exclude that problem is in your PC/serial port

3. Try change cable with a better one. This can exclude the problem is on the cable.

If 1,2,3, fail the problem is in your arduino nano board, buy another one

StuartB4, 2018-09-20:
I noticed in this part of the Session log it mentions 2 different com ports, 3 and 6. 
Just wondered if that might have anything to do with it.

> 09/19/2018 11:38:24 ب.ظ	CloseCom    	Close COM3 [CORE]
> 09/19/2018 11:38:24 ب.ظ	ThreadTX    	System.UnauthorizedAccessException: Access to the port is denied.
> 09/19/2018 11:38:24 ب.ظ	ThreadTX    	   at System.IO.Ports.InternalResources.WinIOError(Int32 errorCode, String str)
> 09/19/2018 11:38:24 ب.ظ	ThreadTX    	   at System.IO.Ports.SerialStream.DiscardOutBuffer()
> 09/19/2018 11:38:24 ب.ظ	ThreadTX    	   at LaserGRBL.ComWrapper.UsbSerial.Close(Boolean auto)
> 09/19/2018 11:38:24 ب.ظ	ThreadTX    	   at LaserGRBL.GrblCore.OnConnectTimeout()
> 09/19/2018 11:38:24 ب.ظ	ThreadTX    	   at LaserGRBL.GrblCore.ThreadTX()
> 09/19/2018 11:38:24 ب.ظ	CloseCom    	System.UnauthorizedAccessException: Access to the port is denied.
> 09/19/2018 11:38:24 ب.ظ	CloseCom    	   at System.IO.Ports.InternalResources.WinIOError(Int32 errorCode, String str)
> 09/19/2018 11:38:24 ب.ظ	CloseCom    	   at System.IO.Ports.SerialStream.DiscardOutBuffer()
> 09/19/2018 11:38:24 ب.ظ	CloseCom    	   at LaserGRBL.ComWrapper.UsbSerial.Close(Boolean auto)
> 09/19/2018 11:38:24 ب.ظ	CloseCom    	   at LaserGRBL.GrblCore.CloseCom(Boolean user)
> 09/19/2018 11:44:26 ب.ظ	Program     	------- LaserGRBL v2.8.36 START -------
> 09/19/2018 11:44:28 ب.ظ	SetStatus   	Machine status [Disconnected]
> 09/19/2018 11:44:34 ب.ظ	Program     	------------ PROGRAM STOP ------------
> 
> 09/19/2018 11:44:48 ب.ظ	Program     	------- LaserGRBL v2.8.36 START -------
> 09/19/2018 11:44:50 ب.ظ	SetStatus   	Machine status [Disconnected]
> 09/19/2018 11:44:52 ب.ظ	SetStatus   	Machine status [Connecting]
> 09/19/2018 11:44:52 ب.ظ	OpenCom     	Open COM6 @ 115200 baud

Also this is mentioned a few times.

> 09/19/2018 11:45:15 ب.ظ	RealTimeStatus	System.FormatException: Input string was not in a correct format.

razamatraz, 2018-10-15:
I haven't gone through all of the steps presented here so I might not be adding anything to the conversation but I have a similar problem with LaserGRBL and a Woodpecker based 3018 machine.

On connection attempt it fails and one (or more) of the steppers go into a vibration that sounds like a locked rotor or pulsing in both directions.  It never stops until I connect with another program. My laser does not fire (unlike other people's issue) though.

This problem existed in GRBL 0.9j and after flashing it with the arduino bootloader and then GRBL 1.1f.  At one point it did work with 0.9j roughly 1/3 of the time; it never works with 1.1f.

Every other program I use (not that I use that many) to connect works:
Candle,
UGS old and new versions
litefire

Creating GCode in LaserGRBL and running it in UGS works too.

davrosko, 2019-02-06:
Hi guys, have you solved that. I am experiencing this issue for a couple of days also. Candle works just fine but when I start laserGRBL all go crazy :( Interesting is that I have used that SW before without any problem. Thank you for any idea how to solve this

EBEALIEN, 2019-03-14:
Hi I bought cnc 3018 laser 5500mw.
I use "grbl candle" and it is good for cnc, for laser I use only "lasergrbl", but when I connect, it starts to burn on the spot or it moves without end and without any g-code. Can you help me to understand whether to return the item or fix it? It may not be the software, but my machine. Thank you

arkypita, 2019-03-14:
@EBEALIEN  #603

StuartB4, 2019-03-14:
I noticed a slight problem in the past when using a different program just before using LaserGRBL.
If I used another program and did not make sure all the axis settings were zeroed befor exiting, when I started LaserGRBL the little blue cross would appear in the position it was in the previous program.
I soon realised what had happened and made sure to zero all settings before shutting the program down and starting LaserGRBL.

EBEALIEN, 2019-03-15:
> 
> 
> I noticed a slight problem in the past when using a different program just before using LaserGRBL.
> If I used another program and did not make sure all the axis settings were zeroed before exiting, when I started LaserGRBL the little blue cross would appear in the position it was in the previous program.
> I soon realised what had happened and made sure to zero all settings before shutting the program down and starting LaserGRBL.

Thank you very much!
Unfortunately I only used lasergrbl version 2.9 included on the disc, and then the last release.
But as soon as I connect the usb and then I click connect, begins to retract the path, up to block the engines by burning continuously with the laser and without g-code.
I removed the 3 sideburns behind, I have uninstalled and reinstalled everything but nothing.
I am desperate!

EBEALIEN, 2019-03-15:
@StuartB4 Forgive me,
I am Italian and I am translating everything indicated by letter every letter. I only understood the message now.
In fact I also used laser web and 2 other software
(I see this tonight, I'm in the office now) I moved the aces and then I went out without throwing any code.
Then entering a code made by me, and using lasergrbl I started without resetting. Tonight I'll try everything well again.
The good thing is that lasergrbl is great and I want to be able to use it, and you all attend very well. Sure we can fix up !!!

StuartB4, 2019-03-15:
Hi  EBEALIEN.
No problem. Translating can be a bit strange somtimes.
I hope you get you machine working properly.
I agree, LaserGRBL is a great program and the support is also excellent.
I know Arkypita the developer of LaserGRBL is always busy, but he is really helpful and spends a lot of his time to make it the best it can be.

arkypita, 2019-03-15:
Have you read the problem I have linked to you #603 and seen my first answer about soft/hard reset?
Did you do any tests changing those values? With what result?

EBEALIEN, 2019-03-16:
Hi Arkypita, report!
_Lap top Win 7 - cnc 3018 5500mW - lasergrbl 3.0.9 - test with file N0001.gcode__ (attached)
Turn on laptop
Cnc3018 laser ignition voltage
connection to laptop with double usb
opening "lasergrbl" - open file "N0001.gcode" - connect USB
I select port com and then connect button of the program
Manual axis movement test OK
"Electronic card" button with laser fan on PWM (has "on" pressed or "PWM" released
Reset and positioning OK - Run execution OK

He executed the program starting from laser off, then it would light up where it was needed and go out where it was not needed. REGULAR. Fast time but the laser light was very weak and did not produce anything, but I don't know if it was the code or my laser.
_Later, I tried my own code, that I don't know how I did it (I'm not able and I used cbride I think a program for cnc engraving) and the laser burned everything, so the laser is fine._

Then turning on the PC and without using other programs before, the axes were stopped and the laser was switched off at power up and machine connection, and, beyond the file whether good or not, the machine moved exactly.

Attached files for control. now I am looking for something to generate them. I have not yet figured out whether to select M3 or M4 and then I also tried with Inkscape the extensions "raste2 laser g-code generation" and "J-thec photonic laser tool", and also the power I don't know if 0 or 255.

Now let's see if we can solve a file from zero to solution, in fact if someone gives me a test to try, I'll try it!
First constructive and positive outcome, lasergrbl must be supported, this site must be supported, the real designers are there. Tell me the donation links.
_Thanks to my dear friend Leonardo_

EBEALIEN, 2019-03-17:
Nothing to do,
Today I launched a file and everything is fine, then I raise the same file,
and the laser illuminates and burns nothing.
I am 7 hours trying not to throw the car out of the window ....

brakthehun, 2019-03-17:
What is your $30 set at??
What diode do you have??

EBEALIEN, 2019-03-17:
> 
> 
> What is your $30 set at??
> What diode do you have??

Good evening, thanks!
I have the cnc 3018 2 in 1 and lasergrbl 3.0.9
My Laser is 450nm and 5500mW, he worked producing work on a piece of wood, even though he didn't work well. Then I recharged but there is neither towards lap top nor on PC, now it is only light. Stop does not burn. How do I try a test to see if the laser works or not?
Thanks for your interest!

EBEALIEN, 2019-04-11:
Hi everyone.
**to inform:**
I returned the cnc 3018 2 in 1 and laser from 5500 mw. Refunded!
Now the cnc 3018 2 in 1 and laser has come from another manufacturer  from 10,000 mv 10 w, with updated card. I will keep you informed.
Someone has info on 10 w and cnc 3018,
how can I make a good test as lightweight configuration LASERGRBL?
[image]
[image]