arkypita/LaserGRBL #151: Compatibility doubt

LuchoGuzman, 2017-10-31:
Hello I would like to know if the software works with the mini laser eleksmaker A5 that works with Benbox if so I would greatly appreciate if you guide me with the installation, since the benbox software is very basic for what I need to do.

arkypita, 2017-10-31:
BenBox software on your PC work only with BenBox firmware on your board and vice versa.
So LaserGRBL doesn't work with a engraver flashed with BenBox firmware.

However you can flash a standard Grbl v0.9 or v1.1 firmware. You lose the ability to use BenBox but you gain the ability to use all the Grbl compatible software like Universal-GCode-Sender, Grbl-controller, jedimaster, ugsplatform, grblpanel, Candle, LaserWeb4 etc

Search/ask to [eleksmaker forum](http://forum.eleksmaker.com/category/6/elekslaser) if someone already do that and maybe they can support/guide you

closed 2017-10-31 by arkypita

LuchoGuzman, 2017-10-31:
If there I could see a video of how to flash the firmware, thank you very much for your response, I appreciate your predisposition very much to the doubts that there are on the part of the users

fasteddyD, 2017-11-06:
I am using a DIY laser that came with the software that strangley resembled Elks Maker software.  I was having issues because the version that came with my laser was in Chinese with a button to convert to English and every so often it would simply revert back to Chinese.  I downloaded Benbox EM for Elksmaker and it worked perfectly in English on my system but still contained the same flaws in other areas, where it generated error codes, but at least they were in English.  Then I found this software and also downloaded it.  It found my laser and works way better than EM, especially for photos!  I did not need to flash the board and can switch back and forth between the two.  My only problem is that the lasergrbl wanders during buring, first left and then right.  It also does not repeat so if you used a double pass the images do not repeat and I get two of them.  I am still attempting to fix this issue because Lasergrbl is the king for what I am attempting to do.  So much so that I will do what every needed to make this work for me, including a board change if necessary.  You will love this software and the tam here are really into what they are doing.  Folks should remember to make a donation once satisfied as these guys work harder and are more responsive than any software developer I have ever been associated with.