arkypita/LaserGRBL #154: Unstable work with diagonal engraving

Mikalay31, 2017-11-05:
1. Horizontal engraving. The engraving speed (1500) is equal to the idle speed (1500). Engraving is stable.
Here are the links to the video:
https://drive.google.com/open?id=1T8qcvb1o3Aj9f_7xp6GtnCEkLvOcNZsN

2. Diagonal engraving. The engraving speed (1500) is equal to the idle speed (1500). There are jerks (apparently there is a slowdown at the time of engraving) although this should not be.
Here are the links to the video:
https://drive.google.com/open?id=1-1i1WEHaZ3sQTVK09qi8yxg--pRBOyxz

Mikalay31, 2017-11-05:
To understand the difference - you need to listen carefully to the sound.

arkypita, 2017-11-06:
Hi @Mikalay31 
I develop LaserGRBL thinking that G0 (fast movement) and G1 (linear movement) could use two different speeds, so I decide to use G0 for white spaces and G1 for engraving.

Later I discover that [G0 has no feed rate](https://github.com/gnea/grbl/issues/279), so any F value used in G0 command is useless. This made the input of idle speed not meaning to the result.

So... at what speed should be executed a G0 command??? The answer is in your Grbl settings that you could show sending $$ command.

Parameters [$110, $111, $110](https://github.com/gnea/grbl/wiki/Grbl-v1.1-Configuration#110-111-and-112--xyz-max-rate-mmmin) control the speed used in G0 and parameters [$120, $121, $122](https://github.com/gnea/grbl/wiki/Grbl-v1.1-Configuration#120-121-122--xyz-acceleration-mmsec2) controls the acceleration.

In my opinion you may have an incoerent seteup of this values between X and Y axis

Mikalay31, 2017-11-06:
Thank you! I understood your answer. I know about the $$ command. But there in the first video you can see that all the parameters are the same. $ 110 = $ 111 = 1500. Therefore I asked you. Can eat still nuances? Simply for engraving the machine with screws it is desirable to work with a constant speed of engraving and idling. And I oche need diagonal engraving. I can reset the pattern of my drawings. You can try on your engraver putting all the parameters the same. I would not be surprised that you will have the same thing. Also did not understand as acceleration on it influences? By the way $ 120 = $ 121 = 300 is also seen in the video. Do you understand me ?

Mikalay31, 2017-11-06:
Here are the settings for $$   https://prnt.sc/h6tobi

Mikalay31, 2017-11-06:
By the way, work at the same speed of engraving and idle speed gives less vibrations for the machine, which increases the quality of engraving. This can be seen on the video with a horizontal engraving. But it increases the processing time. With diagonal engraving, clearly there is a problem. But I'm more interested in quality) Do you understand me?

Mikalay31, 2017-11-06:
And there is one more question. What is the principle of laser power adjustment? What is the frequency of the signal? I need this in order to find the driver specifically for your program. Because not every driver is ready to work at a frequency above 1000 Hertz. In advance I thank for answers. Nicholas.

arkypita, 2017-11-06:
Never eard about google?

> What is the principle of laser power adjustment?
http://www.google.it/search?q=pulse+width+modulation
https://blog.protoneer.co.nz/grbls-pwm-functionality-how-to-control-a-laser-or-variable-speed-spindle/

> What is the frequency of the signal?
https://github.com/gnea/grbl/wiki/Grbl-v1.1-Laser-Mode#laser-mode-operation
https://github.com/gnea/grbl/issues/55

Please note that this functions regards Grbl and they are not dependent on LaserGRBL.

Mikalay31, 2017-11-06:
OK.Thank you.
I understood - I study links ... But on the Internet there is no answer to my main question on diagonal engraving. Help me please )

arkypita, 2017-11-06:
When i have time

Mikalay31, 2017-11-13:
Yesterday once again put the experiments, but already on the latest version v2.8.15. The problem with diagonal engraving remains. Then I changed the background of the circles: Screenshot_6

arkypita, 2017-11-13:
You are asking me to solve a problem that is in your hardware or in your configuration, not a problem that is in LaserGRBL. Do you want to tell me how can I help you?

I have repeat your test with my engraver. As you can see in the linked videos results are pefect both in [horizzontal](https://youtu.be/HKn7CwQx7ok) and [diagonal](https://youtu.be/V2TqOetk0sk) mode.

LaserGRBL optimize the speed of the engraving sending G0 (fast movement) to skip all the whitespaces.
G0 is implemented by Grbl moving the laser at the faster speed possible, as settled in parameters [$110, $111, $112](https://github.com/gnea/grbl/wiki/Grbl-v1.1-Configuration#110-111-and-112--xyz-max-rate-mmmin) and parameters [$120, $121, $122](https://github.com/gnea/grbl/wiki/Grbl-v1.1-Configuration#120-121-122--xyz-acceleration-mmsec2).

Mikalay31, 2017-11-13:
OK. I understood. I will test or just use horizontal engraving. Tell me another question. There are buttons in the program, but I do not really understand what to use them for. Where can I read about this? I will attach a photo with questions:   
[image]        
[image]    
[image] 

Thank you in advance !

Mikalay31, 2017-11-13:
What other buttons can I still insert into the program?

arkypita, 2017-11-13:
http://lasergrbl.com/it/usage/overrides/
http://lasergrbl.com/it/usage/user-interface/
http://lasergrbl.com/it/configuration/#streaming-mode
http://lasergrbl.com/it/usage/custom-buttons/

arkypita, 2017-11-13:
issue detector and unidirectional engraving are very new feature and I have not already documented but you can read about that in github releases storyboard

https://github.com/arkypita/LaserGRBL/releases

Mikalay31, 2017-11-15:
Hey. I solved my problem) The reason is not in your program and not even in my hardware. The problem is probably in the GRBL itself. It is necessary to set the speed according to my copies from the screen. All that I can assume lies in the formula of the hypotenuse of the triangle. Moving along the X, Y axis at a speed of 1200 mm / min - will cause diagonal movement at a speed of 1697 mm / min. But this is only an assumption. Probably that that is not finished in calculation G0.

Mikalay31, 2017-11-15:
[image]
[image]

Mikalay31, 2017-11-15:
[image]
[image]

arkypita, 2017-11-15:
Please note again:
> I discover that G0 has no feed rate, so any F value used in G0 command is useless. This made the input of idle speed not meaning to the result.

So: **the second speed parameter (the one you set at value of 1200) has no effect at all.**

The speed of G0 used in whitespaces is implemented by Grbl moving the laser at the faster speed possible, as settled in parameters $110, $111, $112 and parameters $120, $121, $122.

Mikalay31, 2017-11-16:
I understood. Thanks for the help ! Great program! I will be watching your project. He is very interesting !

closed 2017-11-16 by arkypita