arkypita/LaserGRBL #155: v2.8.11 need testing

arkypita, 2017-11-05:
Hi all, I have released a new [version v2.8.11](https://github.com/arkypita/LaserGRBL/releases/tag/v2.8.11) that need testing.
A feature has been added that intercepts possible grbl hang/issue during job streaming and show a dialog for to warn user about this issue.

This intercept could detect:
- grbl stop responding to ? (No Real-time Status Reports) for more then 10s
- no movement when a movement is requested for more then 10s
- unexpected grbl reset during a job

I need some testing to be sure that this intercept does not trigger by mistake.
If you decide to test this version please reply to this message.

Mikalay31, 2017-11-06:
Hey. I'm ready )
By the way, I can help. I had such a problem.
I decided it) At me during work at me the machine stopped.
Yesterday I tested everything was normal.

Mikalay31, 2017-11-08:
Today the update window has been opened: http://prntscr.com/h7xac0     Is this the latest version?

arkypita, 2017-11-08:
Yes! I decide i am sure about new features so i made them available for update

arkypita, 2017-11-08:
p.s: alt+stamp and ctrl+v allow to paste image directly in github message

closed 2017-11-09 by arkypita