arkypita/LaserGRBL #157: Wish for New feature :) Direct Goto to Dialog "Target Image"

midibase, 2017-11-07:
As the title say, I wish a commat where I can got direct to the dialog-window "Target image".
The reason is that I must often change the values for best results.

Now I must always use File->Open File.. and choose always the same image.

Or is now a possibility?

Thanks for your great work!
lg sigi

arkypita, 2017-11-07:
Just only to save one click? Open file always remember the last opened file, so you don't need do navigate, just confirm to open the same image.

midibase, 2017-11-08:
Yes thats right. It is no Problem on a PC where you have a mouse.

But on my solution in the shop I use a Laptop without mouse (risen is a placeproblem)

So each movement on the laptop is very tricky. So it were be very usefall, If I could do a Command on a button or has a shortkey for the command.

 

BG

 

Sigi

StuartB4, 2017-11-08:
You could just use a USB mouse to plug in to the laptop.

arkypita, 2017-11-08:
@StuartB4 he say it is using a Laptop without mouse because of placeproblem
@midibase it's ok for me to add a File menu entry with "reopen last file" as well as add some keyboard shortcut for jogging as already asked by someone

midibase, 2017-11-08:
The problem is not cable or wireless. The problem is hat i built a cool station, where place on one level for the Laser is and

on the over level is place only for the laptop. No place for a mouse ☹ on this level.

 

 

Bg

 

sigi

StuartB4, 2017-11-08:
Oh I see, ok.

closed 2017-11-16 by arkypita

arkypita, 2017-11-18:
added in [v2.8.17](https://github.com/arkypita/LaserGRBL/releases/tag/v2.8.17)
please test and comment #171

midibase, 2017-11-18:
Hey cool,

now I am going to test it, I tell you thgen my impression

 

bg

 

sigi

midibase, 2017-11-18:
Hello

I tested and it works well. But some impressions or question do I have

*	It is not clear for me how I could use the CustomButtons – short cuts. 
I do not found a way where I could  link it for example the custum made buttons.
*	Not really nessecary but a shortcut could also shown on the menuItem for example Strg+O for opening File
*	I’m not sure if it is good or it is nessesary to add new shortCuts (I would it not needed)?
*	Also a little thing is, if the Buttons on a Dialog (OK, Next…) would have shortcuts

 

Some other questions:

*	For what is homing ($H) really needed?
*	A really cool feature where, if it were possible to „draw“ the lines-light. 
This would mean in the option to set the s-max value (f.e. 20). 
And when it start it  moves with the maximum speed (G0) and with the laser-power from the option. So I could show if all drawings are right on the wood.
I’m also an c# programmer, so if it is ok for you  I would build this function, and when you think it is good enough, so you could put the code in your main-code?

 

 

 

 

Bg

 

Sigi

arkypita, 2017-11-18:
> It is not clear for me how I could use the CustomButtons – short cuts. I do not found a way where I could  link it for example the custum made buttons.

Linking is positional: CustomButton1 is for the first button, CustomButton2 for the second...

> I'm not sure if it is good or it is nessesary to add new shortCuts (I would it not needed)?

I think I have added all the available actions to the shortcuts list. Tell me if you see I forget something.

> Also a little thing is, if the Buttons on a Dialog (OK, Next…) would have shortcuts.

They usually have standard windows shortcuts (Esc: cancel, Enter: ok) but sometime I forget to define them. Some other times I choose to not define them because in some situation an "explicit" click is more clear than an enter (think about a form where you have an edit and you click ok thinking you are editing a value and an ok is taken on the form instead!).

> For what is homing ($H) really needed?
Homing is a procedure to get a reliable and absolute position reference sensing limit switch.
It require hardware limit switch: https://www.youtube.com/watch?v=ZyS-SgZtbXQ

> A really cool feature where, if it were possible to „draw“ the lines-light.

In LaserGRBL project I choose to prefer simplicity to "a lot of cool features that each one would want a bit different, driving me crazy". For positioning you can use a custom button and draw image frame.