arkypita/LaserGRBL #159: Problem with the laser

LuchoGuzman, 2017-11-08:
Hello, I would like to know if you have any idea why when I load the grbl firmware, I leave the laser burning all the time, I work with an eleksmaker A3 pro, I used your software beforehand and it was very useful compared to Bembox, which is the one that came to me for a recommendation with the team, but since yesterday I do not know what was changed that the laser burns me all the time when I use laser grbl

LuchoGuzman, 2017-11-09:
clarified that when I use other software, the laser works correctly, but when loading grbl v1.1 firmware and using the laser grbl software, the laser remains burning all the time

arkypita, 2017-11-09:
M3 turn on laser, M5 turn off
on and off are managed by Grbl firmware inside arduino. There is **nothing in LaserGRBL** that can match your problem.

If you send this M3, M5 command manually and the laser still remain on it mean you have connected the laser or ttl control to the wrong arduino pin.

Check grbl documentation https://github.com/gnea/grbl/wiki/Connecting-Grbl

closed 2017-11-09 by arkypita