arkypita/LaserGRBL #160: Can't enter negative offset

MikeLink83, 2017-11-09:
I have a router and laser mounted together and have gotten into the habit of centering my XY origin at the center of my work piece to make alignment easier between my outputs from Fusion 360 (Facing/drilling/cutout operations) and ArtCAM (3d reliefs).  I've figured out how to apply my offsets and pre-position my machine before switching to the Arduino controller so X0 Y0 are at the lower corner of the artwork I want engraved in LaserGRBL but it is a very exacting process that is easy to mess up by getting a positive/negative offset backwards.
Can we please set for example:
Image size W:400 H:300
Offset  X:-200 Y:-150

Amazing program by the way 👍 

Example of project using 3 different programs and 2 controllers that requires fairly precise alignment
[image]

My solution for laser/spindle shared lift.
[image]

brakthehun, 2017-11-09:
Another thing in this same categorize is I can't make my line density less than 1mm. When I pull the focusing lense off my CO2 laser,, the beam is 2.5mm.

arkypita, 2017-11-09:
ouch... i never think that someone needs line density less then 1mm :-)

arkypita, 2017-11-16:
added to development roadmap #64

closed 2017-11-16 by arkypita

arkypita, 2017-11-21:
v2.8.20 implement negative offset
https://github.com/arkypita/LaserGRBL/releases/tag/v2.8.20

MikeLink83, 2017-11-21:
Very nice. I tried it out already and it's exactly what I was after. Now I
can keep my controllers zeroed out on the center of the work piece.

Thank you