arkypita/LaserGRBL #161: v2.8.13 need testing (again)

arkypita, 2017-11-09:
Hi all! I have released a new version that can handle a "resume from last command" functionality that allow user to recover form an interruption (hang, disconnection...). 

When you press "play" on a file that streaming was previously stopped, it detect the situation and gives the chance to resume from previous position. I have developed this function without the presumption that he can do miracles, but if it work it could made the difference.

Of course you must not exit from LaserGRBL, manually move your engraver head, or load another file, but if you want you can do $H Homing.

I have tested it with vector and absolute positions.
Not tested at all with raster and relative coordinates.

[[image]](https://https://www.youtube.com/watch?v=UwJAFp1pS-I)

arkypita, 2017-11-09:
Already known issue: the position I suggest as "last command managed" is the one that receive the last "OK" response. It is good but it is some step forward against the real execution because of planned buffer (the ok from Grbl mean "i put in planned buffer", not "I have executed it"), so it skip some lines.

Mikalay31, 2017-11-13:
I tested the version v2.8.15. Works great!

closed 2017-11-16 by arkypita

ChristopheVersieux, 2023-11-09:
Hello
Is there an option to start in the middle of a design? Laptop went to sleep and I couldn't reconnect. I had to restart my design and would like to skip a part of it as it is already burned...

Mikalay31, 2023-11-09:
Hello.
Honestly... I haven't used it for a long time. The program is excellent.
As far as I remember, this was possible - it depends on the version you
installed.

I don't want to deceive. The program is good. But the program is still a
little outdated...alas.

ChristopheVersieux, 2023-11-09:
@Mikalay31 I haven't seen such an option on the latest version I installed before starting my project

arkypita, 2023-11-09:
menu file, send from position