arkypita/LaserGRBL #165: LaserGRBL as UV PCB printer

emilvv, 2017-11-16:
Hello,

First - Thank you for your great work !!

It is very interesting to use your software to control UV PCB printer !

I have a question ..... please help to choose proper setup ...

I tried many setup options, but received strange processing time - it is not normal ...
please see "11 hours 3 min.jpg" .... so ..

The base information:

- CNC machine / printer with timing belts and microstepping motor drivers - "CNC.jpg"

- UV laser diode with PWM control electronics ....

IMPORTANT ! - laser beam is focused on about 0.04 mm ... so lines have to be 0.04 mm or 0.05 mm !!

Example pcb image to plot / print on UV resist on PCB - "PCB.jpg"

GRBL version - 1.1f ... ,   LaserGRBL version 2.8.12 ....

My question is:

How to setup LaserGRBL - "setup.jpg" and to have reasonable working / engraving time ..

as this video ...https://www.youtube.com/watch?v=Uk2fGoNL3Yk

not as my example: "11 hours 3 min.jpg"

Thank you for your help,

Emil - emilvv@gmail.com

[image]
[image]
[image]
[image]

arkypita, 2017-11-16:
20 lines/mm in 70mm x 70mm means a total linear line development of 70x20x70 = 98000mm
At F4000 the computed (estimated) time should be about 98000/4000 = 25min

When you begin engraving, LaserGRBL compute a "projected time" based on the current REAL speed of your engraver. So if you set F4000 (4000mm/min) but your engraver execute the job slower the "projected time" recompute the time runtime and "on the fly" to match the real execution speed.

arkypita, 2017-11-16:
If you type $$ in the command box "type gcode here" you can see the configuration params settled on your hardware. They could limit your speed because the speed will never exceed params in $110 $111 $112 even if you ask F100000000 from LaserGRBL

[image]

emilvv, 2017-11-16:
THANK YOU !!!!

My $110 $111 $112 = 500.000 ...

1. If I change $110 $111 $112 to 7000 it will be better ? .. How to change them - in GRBL ?? 

2. May I ask - can LaserGRBL work with constant speed ( raster plotting  ) AND with laser just ON and OFF ?

If you plot UV on PCB it is important ... and one more question - if we plot a PCB tracks image line by line - raster plot .. how about end lines acceleration  ?

Best regards,

Emil

arkypita, 2017-11-16:
1. You can change them, but only if physical hardware support this speed. Type $num=value in command box (ie $110=7000) to write grbl config.

If you wait a minute i am releasing a new version with a simplified configuration editor.

2. M3 mode is fixed power mode. M4 is dynamic power mode. read: https://github.com/gnea/grbl/wiki/Grbl-v1.1-Laser-Mode

emilvv, 2017-11-16:
Thank you again !!

I made it .. and also $31=1 - laser mode .... great .. time is perfect - 20 min !!! 

( simulation only on my PC - without real machine ... waiting for some components !! ) 

Have a nice day !!

Emil

arkypita, 2017-11-16:
I look forward to seeing the results of this work
Best regards, Diego

closed 2017-11-16 by arkypita