arkypita/LaserGRBL #166: Version 2.8.16 Error when writing changes to GRBL settings

StuartB4, 2017-11-16:
Hi. Just updated to 2.8.16 and tried to change the settings for $110 and $111.
When writing the changes to GRBL I get the error in this image, even though I never changed $32.

[image]

Changing $32 and writing to GRBL also produces the same error.

arkypita, 2017-11-16:
LaserGRBL send the whole config set, not only changed values.
If you try to change $32 manually by typing $32=0 inside "type gcode here" box, do you have the same error?

StuartB4, 2017-11-16:
I think it has something to do with PWM. My laser is not using PWM at the minute, just on and off.
So I had a look at GRBL error 17 and it says:
"17 | Laser mode requires PWM output."

arkypita, 2017-11-16:
Yes. i think that PWM is disabled in hardware ([VARIABLE_SPINDLE in config.h](https://github.com/gnea/grbl/blob/5967839ab38f8e984a384db1f64e781cffb9e739/grbl/config.h#L339) of grbl) and grbl reply with this error at each try to set $32

arkypita, 2017-11-16:
It would probably be better if I limit writing only to the edited values

StuartB4, 2017-11-16:
Yes, my config.h file has the 2 // in front of the setting on line 339.
When I get a new laser driver board that supports PWM I will remove the 2 //'s
to enable it properly.
Thanks very much for the help.

arkypita, 2017-11-18:
fixed in [v2.8.17](https://github.com/arkypita/LaserGRBL/releases/tag/v2.8.17)

closed 2017-11-18 by arkypita

StuartB4, 2017-11-19:
Thats great, thanks very much.