arkypita/LaserGRBL #168: I can not download the executable

cecconilorenzo, 2017-11-17:
the antivirus locks it because it contains FileRepMalware

arkypita, 2017-11-17:
Already known #70 

The false positive from AV heuristic filter is a know problem. Unlikely many AV consider suspicious any sw without digital signature. Also the fact i use self extracting .zip trigger some AV. I have tried with others installer but the problem still exist. Is something I can not control. If anyone have suggestions :-)
You trust me? You can whitelist and install LaserGRBL.

Ciao

cecconilorenzo, 2017-11-17:
ok, grazie per il lavoro straordinario che fai.

arkypita, 2017-11-17:
prego

closed 2017-11-17 by arkypita

StuartB4, 2017-11-18:
Which antivirus software are you using. 
Most of them allow you to make a choice to exclude a file from the scan if you know it is safe to use.
You should be able to unlock the file and exclude it from further scans or detection by the antivirus program.

arkypita, 2017-11-21:
@cecconilorenzo 
Whould you like to test last version? I have changed my installer and now should be more antivirus friendly.
https://github.com/arkypita/LaserGRBL/releases/tag/v2.8.20