arkypita/LaserGRBL #170: Jagged lines while creating g-code

hk89, 2017-11-17:
Hi,

I figured out that LaserGRBL does not create arcs and curves for most of the vectorized picture. Sometimes it appears that most curves are somehow turned into stair-stepped lines which wastes time, increases g-code length and distance traveled. It appears on the g-code preview screen. However they do not appear on the screen where I adjust the settings for it. I tried smoothing and downsampling without a good result. I even changed the junction deviation from 0.01 to 0.001

Please see pictures and gcode

[image]

[image]

[gcode.zip](https://github.com/arkypita/LaserGRBL/files/1484040/gcode.zip)

arkypita, 2017-11-17:
Use Smooth instead of Sharp as resize option.

arkypita, 2017-11-17:
and read usage guide: http://lasergrbl.com/it/usage/raster-image-import/import-parameters/

hk89, 2017-11-17:
Thanks for the kind reply. I changed it now and it is working like a charm. Sorry for wasting your time. I guess I did not see that one in the usage guide.

closed 2017-11-17 by hk89