arkypita/LaserGRBL #171: v2.8.17 need testing

arkypita, 2017-11-18:
I have added keyboard shortcut managment. Please someone test it before I release as official.
https://github.com/arkypita/LaserGRBL/releases/tag/v2.8.17

[image]

arkypita, 2017-11-18:
I also add a "Set zero point" button, because a lot of user is asking for it.
I also made the "Homing" button appare/disappear if configured in Grbl

TehseenHasan, 2017-11-18:
That's very nice....

arkypita, 2017-11-18:
released, hope it work without sideeffects

StuartB4, 2017-11-19:
Hi Arkypita.
I cant get the "Set zero point" button to work in the latest version.
I move the X or Y axis a few millimeters and the press the "Set zero point" button and the blue + jumps straight back to the bottom left home position.
It also affected my custom button which did work to set a new X0 Y0 position, that now does the same and jumps straight back to the bottom left home position when clicked.

StuartB4, 2017-11-19:
Just realised, it does store the new 0,0 position but only shows after closing the program, restarting and clicking on Connect. Then the blue +jumps to where I moved the X,Y axiz to.

aMirMOsavi, 2017-11-19:
Hi
get error While Use Keyboard shortcut !
[image]
[error Keyboard shortcut.txt](https://github.com/arkypita/LaserGRBL/files/1485462/error.Keyboard.shortcut.txt)

arkypita, 2017-11-19:
> I cant get the "Set zero point" button to work in the latest version.
I move the X or Y axis a few millimeters and the press the "Set zero point" button and the blue + jumps straight back to the bottom left home position.
It also affected my custom button which did work to set a new X0 Y0 position, that now does the same and jumps straight back to the bottom left home position when clicked.

@StuartB4 
It is correct, it is the desired behavior by design.
If you set a new zero, the blue cross is now at zero. Does this make sense?

The error was in previous version of LaserGRBL that still showing the cross in absolute position. 
In my opinion now is better.

arkypita, 2017-11-19:
@aMirMOsavi 
What are the conditions for reproducing the problem?

StuartB4, 2017-11-20:
> It is correct, it is the desired behavior by design.
If you set a new zero, the blue cross should go at zero. Does this make sense?

> The error was in previous version of LaserGRBL that still showing the cross in absolute position.
In my opinion now is better.

Yes I understand now, thanks very much.

arkypita, 2017-11-20:
@aMirMOsavi 
I have googled about that, seems to be a bug of Microsoft DataGridView control. References:
https://stackoverflow.com/questions/34344499/invalidoperationexception-this-operation-cannot-be-performed-while-an-auto-fill
https://connect.microsoft.com/VisualStudio/feedback/details/481029/this-operation-cannot-be-performed-while-an-auto-filled-column-is-being-resized

I have implemented a workaround for future versions.

StuartB4, 2017-11-20:
Hi Arkypita.
I just had a look at the error text file aMirMOsavi supplied and noticed this:

> ************** Loaded Assemblies **************
mscorlib
    Assembly Version: 2.0.0.0
    Win32 Version: 2.0.50727.8825 (WinRelRS3.050727-8800)
    CodeBase: file:///C:/Windows/**Microsoft.NET/Framework64/v2.0.5072**7/mscorlib.dll

Just wondered if that could be a problem as it's a lower version than you recommend for running LaserGRBL.

arkypita, 2017-11-20:
Good observation!

arkypita, 2017-11-20:
In v2.8.19 I add a fix for @aMirMOsavi issue
I also have changed the update procedure that allow me to use a different installer/packager (instead of packing with winrar self extractor) that should be more antivirus friendly.

closed 2017-11-21 by arkypita