arkypita/LaserGRBL #173: Use as Cutter too?

brucek5, 2017-11-19:
Your program is great, and I know it's focus is on image engraving. I like that it's a "one stop shop" app. I use my laser for cutting "Laser Board" to make small model parts.  I tried your program today to do cutting and it worked, but I had to make the overall image file larger to cut the small parts. I used your Vectorization mode which kind of worked, but because it "traces" (a cut on both sides of a line) the cut size was off.  Maybe in the future you could add a "cut-mode" where you make one cut path based on finding the middle of the existing two trace paths. And allow smaller image sizes. Just a thought, good job!

arkypita, 2017-11-19:
Hi @brucek5 
What you are talking about is called "centerline tracing" and it is already in development roadmap #64 
However it is not so easy for a computer to detect a centerline and result are not always as good as you want.

i.e try: https://online.rapidresizer.com/tracer.php
with: https://upload.wikimedia.org/wikipedia/commons/thumb/f/f3/Drawing_line.svg/1200px-Drawing_line.svg.png

and see the line 6

arkypita, 2017-11-19:
Also you can change your mode to draw pieces: if you want to cut this

[image]

Draw it like this

[image]

Evil-Nemesis, 2017-11-21:
hi there

arkypita, 2017-11-21:
Thanks for the demonstrated interest. Unfortunately it will not be so simple because there is no centerline library for C# available (or I am not able to found)

closed 2017-11-23 by arkypita

arkypita, 2018-10-21:
Hi all!

I have released a new version of LaserGRBL that include **centerline vectorization**.
This feature was requested by a lot of user and complete LaserGRBL with a "must have features" for all who need a simple way to engrave images.

This release was released as a pre-release so it is not automatically updated. 
To those who want to test it I recommend downloading the installer from: [here](https://github.com/arkypita/LaserGRBL/releases/download/v3.0.0/install.exe)

brucek5, 2018-10-22:
arkypita, thanks for the update! I'll be testing it this weekend.

ooYuYoo, 2023-12-04:
Hello, I really admire the work you do. This is my first time commenting on GitHub, and I would be delighted if you could reply. I would like to ask for the website link you provided for online vectorization of centerlines, and could you please share the approach you used for vectorizing centerlines? I am currently very interested in this, but I haven't found any relevant papers or work. When converting SVG to DXF format, I noticed that the DXF file contains the contours of the SVG image, but I want to extract the centerlines. I would appreciate it if you could reply. Thank you once again!
[image]
@arkypita